Xilinx-CNS/onload #193: Issue with Installing Alveo X3522 Net Driver: Module Not Found Error

alitokur, 2023-12-06:
### Environment
- **OS**: Red Hat Enterprise Linux release 8.7
- **Device**: Alveo X3522

### Issue Description
I am encountering a problem while trying to install drivers for the Alveo X3522 device. I downloaded the RPM package (XN-201053-LS Alveo X3522 Net Driver Source) and built it using the following command:

`rpmbuild --rebuild`

The build process completed without any errors. However, when I attempt to run the following command:

`modprobe -r xilinx_efct`

I receive the following error message:

`modprobe: FATAL: Module xilinx_efct not found.`

### Steps to Reproduce
1. Download the RPM package (XN-201053-LS Alveo X3522 Net Driver Source).
2. Run `rpmbuild --rebuild` to build the package.
3. Run `modprobe -r xilinx_efct`.

ivatet-amd, 2023-12-06:
Hello @alitokur,

```
modprobe -r xilinx_efct
```
`-r` tries to unload the driver if it was previously loaded. Is it the case for you?

```
modprobe: FATAL: Module xilinx_efct not found.
```
It looks like `modprobe` can't find `xilinx_efct.ko`. Have you installed the package you built?

I tried to rebuild it and found `xilinx_efct.ko` in the binary RPM file:
```
$ rpmbuild --rebuild xilinx-efct-1.5.5.0-1.src.rpm

$ cd ~/rpmbuild/RPMS/x86_64/

$ ls
kernel-module-xilinx-efct-RHEL8-4.18.0-372.9.1.el8.x86_64-1.5.5.0-1.x86_64.rpm

$ rpm2cpio kernel-module-xilinx-efct-RHEL8-4.18.0-372.9.1.el8.x86_64-1.5.5.0-1.x86_64.rpm | cpio -idmv
./lib/modules/4.18.0-372.9.1.el8.x86_64/build/include
./lib/modules/4.18.0-372.9.1.el8.x86_64/build/include/linux
./lib/modules/4.18.0-372.9.1.el8.x86_64/build/include/linux/net
./lib/modules/4.18.0-372.9.1.el8.x86_64/build/include/linux/net/xilinx
./lib/modules/4.18.0-372.9.1.el8.x86_64/build/include/linux/net/xilinx/xlnx_efct.h
./lib/modules/4.18.0-372.9.1.el8.x86_64/updates/xilinx_efct.ko
1027 blocks
```

Please also note this repository is OpenOnload. For the X3 issues, please email support-nic@amd.com. If this is your first time doing this, you should receive an automated email with the details to sign up.

Often it is useful to include the sfreport. Please run the Solarflare Linux diagnostic script using ‘perl sfreport.pl’ (as root) in your system and send us back the generated HTML report. The latest version of sfreport can be downloaded from the Linux section of https://www.xilinx.com/support/download/nic-software-and-drivers.html#drivers-software.

alitokur, 2023-12-06:
sorry forgot to install the package after building. now its working well.

closed 2023-12-06 by alitokur