Xilinx-CNS/onload #196: onload_version_gen: Trailing space in ONLOAD_SHORT_VERSION causes DKMS incompatibility

nodakai, 2023-12-16:
It produces an `ONLOAD_SHORT_VERSION` definition with a trailing space, as shown here:

```
#define ONLOAD_SHORT_VERSION "v8.1.2 "ONLOAD_BUILD_PROFILE
```
]
This extra space leads to compatibility problems with DKMS, which uses a version string generated by `scripts/lib/mkdist/preprocess_source. As a result, I've got assertion errors like the following:

```
Dec 16 17:26:40 myserver kernel: [onload] ERROR: user/driver version mismatch
Dec 16 17:26:40 myserver kernel: [onload]   user-version: v8.1.2 
Dec 16 17:26:40 myserver kernel: [onload]   driver-version: v8.1.2
Dec 16 17:26:40 myserver kernel: [onload] HINT: Most likely you need to reload the sfc and onload drivers
```

Notice that the `user-version` line has a space at the end, which seems to be coming from `onload_version_gen`.

I haven't set `ONLOAD_BUILD_PROFILE` in my environment, and I'm puzzled as to how this assertion could ever pass if this value were set.

As a temporary fix, I removed the trailing space from `onload_version_gen` and rebuilt the user space. This resolved the assertion issue, and I'm now able to use OpenOnload with SolarFlare NICs successfully.

abower-amd, 2023-12-16:
Thanks for investigating this, @nodakai! Please feel free also to raise a pull request if you find a general solution.

jfeather-amd, 2023-12-18:
Thanks for raising this issue @nodakai! I'm glad you were able to find a workaround for this, and I appreciate your documentation of this issue. As discussed in #145, I have a fix in review for this issue as well, but GitHub closed the issue a little early so I will be reopening until the fix is merged.

closed 2023-12-20 by jfeather-amd