Xilinx-CNS/onload #200: SFC net driver fails to build for linux-6.1.latest

ol-alexandra, 2024-01-23:
I see an issue with the latest updates for linux-6.1 kernel: 6.1.73-2.el9.elrepo.x86_64 from elrepo repository with CentOS9 and 6.1.0-17-amd64_x86_64 with Debian 12.
```
  CC [M]  /home/sasha/work/level5/onload-b/build/x86_64_linux-6.1.73-2.el9.elrepo.x86_64/driver/linux_net/drivers/net/ethernet/sfc/nic.o
In file included from /home/sasha/work/level5/onload-b/build/x86_64_linux-6.1.73-2.el9.elrepo.x86_64/driver/linux_net/drivers/net/ethernet/sfc/driverlink_api.h:16,
                 from /home/sasha/work/level5/onload-b/build/x86_64_linux-6.1.73-2.el9.elrepo.x86_64/driver/linux_net/drivers/net/ethernet/sfc/net_driver.h:86,
                 from /home/sasha/work/level5/onload-b/build/x86_64_linux-6.1.73-2.el9.elrepo.x86_64/driver/linux_net/drivers/net/ethernet/sfc/efx.c:33:
/home/sasha/work/level5/onload-b/build/x86_64_linux-6.1.73-2.el9.elrepo.x86_64/driver/linux_net/drivers/net/ethernet/sfc/filter.h:40: error: "__struct_group" redefined [-Werro]
   40 | #define __struct_group(TAG, NAME, ATTRS, MEMBERS...) \
      | 
In file included from ./include/linux/stddef.h:5,
                 from ./include/uapi/linux/posix_types.h:5,
                 from ./include/uapi/linux/types.h:14,
                 from ./include/linux/types.h:6,
                 from ./include/linux/kasan-checks.h:5,
                 from ./include/asm-generic/rwonce.h:26,
                 from ./arch/x86/include/generated/asm/rwonce.h:1,
                 from ./include/linux/compiler.h:246,
                 from ./include/linux/build_bug.h:5,
                 from ./include/linux/container_of.h:5,
                 from ./include/linux/list.h:5,
                 from ./include/linux/module.h:12,
                 from /home/sasha/work/level5/onload-b/build/x86_64_linux-6.1.73-2.el9.elrepo.x86_64/driver/linux_net/drivers/net/ethernet/sfc/efx.c:11:
./include/uapi/linux/stddef.h:26: note: this is the location of the previous definition
   26 | #define __struct_group(TAG, NAME, ATTRS, MEMBERS...) \
      | 
In file included from /home/sasha/work/level5/onload-b/build/x86_64_linux-6.1.73-2.el9.elrepo.x86_64/driver/linux_net/drivers/net/ethernet/sfc/driverlink_api.h:16,
                 from /home/sasha/work/level5/onload-b/build/x86_64_linux-6.1.73-2.el9.elrepo.x86_64/driver/linux_net/drivers/net/ethernet/sfc/net_driver.h:86,
                 from /home/sasha/work/level5/onload-b/build/x86_64_linux-6.1.73-2.el9.elrepo.x86_64/driver/linux_net/drivers/net/ethernet/sfc/nic.c:20:
/home/sasha/work/level5/onload-b/build/x86_64_linux-6.1.73-2.el9.elrepo.x86_64/driver/linux_net/drivers/net/ethernet/sfc/filter.h:40: error: "__struct_group" redefined [-Werro]
```

ivatet-amd, 2024-01-23:
The fix is already in the Netdriver repo, but it hasn't been tagged yet. Once tested and tagged, I'll update Onload. 👍

ol-alexandra, 2024-01-26:
It's been fixed by now, thank you.

ol-damirm, 2024-01-29:
Unfortunately, I did not find any relevant changes in the repository.
@ivatet-amd Perhaps they haven't been applied yet?

ivatet-amd, 2024-01-29:
The Net driver team tagged the fixed revision last week, and I'm working on updating Onload. There are minor issues, so it won't take long.

ivatet-amd, 2024-01-29:
A quick update: As soon as I merged the new Net driver, it broke CI for an older RHEL. The breakage is trivial, but at the boundaries between Onload and Netdriver, we need to think about a better long-term solution.

The newer Net driver is https://github.com/Xilinx-CNS/onload/commit/6fa0390c8faa6816d59a2fec6103a85eacacac67 if you want to try it against the more recent kernels, but please mind that it's currently reverted with the subsequent commit so that the latest merge 22e08ad6498ae4a1a5eba923763b0b22b9872304 is a no-op.

ivatet-amd, 2024-02-05:
Another quick update: The build-time/interfacing issue is now being addressed. I'm anticipating a new tagged Net driver soon.

ivatet-amd, 2024-02-14:
The updated Net driver `v5_3_18_1005` from the merged https://github.com/Xilinx-CNS/onload/commit/c552f701dc2f4fe5ea44878674d6c817a6ed82e9 should be good, please give it a go.

Meanwhile, I would like to resolve this issue and continue addressing https://github.com/Xilinx-CNS/onload/issues/203.