Xilinx-CNS/onload #207: linux-6.7 + huge pages: Failed to map packets!

okt-sergeyn, 2024-03-01:
Onload fails to create stack on Linux 6.7 with enabled huge pages:
```
$ onload nc -l 20002
oo:ncat[91964]: __ci_netif_pkt: mmap(0x0000000040000000) failed for pkt set 1 (0)
oo:ncat[91964]: Failed to map packets!
oo:ncat[91964]: FAIL at /home/nikitins/work/onload/src/lib/transport/ip/netif.c:1220
oo:ncat[91964]: ci_assert(ef_eplock_is_locked(&(ni)->state->lock))
from /home/nikitins/work/onload/src/lib/transport/ip/netif.c:1220

```
It works with `EF_USE_HUGE_PAGES=0` or in AF_XDP mode.

ivatet-amd, 2024-03-20:
This is caused by https://lore.kernel.org/lkml/20230907001832.GA63356@monkey/T/ where the page index became `PAGE_SIZE`-grained (and not hugepage-size grained).

ivatet-amd, 2024-03-20:
Must be fixed by https://github.com/Xilinx-CNS/onload/commit/d8201b2fc82bd56c73d89e3d16ed70c382fa15ec. 👍

closed 2024-03-20 by ivatet-amd