Xilinx-CNS/onload #208: changing combined channels in the interface does not change the RX flow hash indirection table

ol-damirm, 2024-03-07:
Changing combined channels in the interface does not change the RX flow hash indirection table.
I see this behaviour starting with commit c552f701dc2 (v5_3_18_1005).

Here's how it can be reproduced:
```
# ethtool -i enp2s0f0 |grep version
version: 5.3.18.1005
firmware-version: 8.5.0.1002 rx0 tx0

# ethtool -x enp2s0f0
RX flow hash indirection table for enp2s0f0 with 4 RX ring(s):
    0:      0     0     0     0     0     0     0     0
    8:      0     0     0     0     0     0     0     0
   16:      0     0     0     0     0     0     0     0
...
  120:      0     0     0     0     0     0     0     0
RSS hash key:
00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
RSS hash function:
    toeplitz: on
    xor: off
    crc32: off
```

So I see zeros in the table and in RSS hash key.
After that, I try to change the number of combined channels, it changes, but RX flow hash indirection table values  remain with zeros.
```
# ethtool -l enp2s0f0
Channel parameters for enp2s0f0:
Pre-set maximums:
RX:             n/a
TX:             n/a
Other:          1
Combined:       31
Current hardware settings:
RX:             n/a
TX:             n/a
Other:          1
Combined:       4

# ethtool -L enp2s0f0 combined 2
# ethtool -x enp2s0f0
RX flow hash indirection table for enp2s0f0 with 4 RX ring(s):
    0:      0     0     0     0     0     0     0     0
    8:      0     0     0     0     0     0     0     0
   16:      0     0     0     0     0     0     0     0
...
  120:      0     0     0     0     0     0     0     0
RSS hash key:
00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
RSS hash function:
    toeplitz: on
    xor: off
    crc32: off
```

I use:
- Debian-12
- linux-6.1.0-17-amd64
- Solarflare Communications SFC9120 10G Ethernet Controller [1924:0903] (rev 01)

@ivatet-amd with the previous driver version ( 5.3.16.1004) it  works as expected.

ivatet-amd, 2024-03-18:
Hi @ol-damirm, sorry somehow I missed this issue in my inbox and noticed only now. I will try to take a look this week.

ol-alexandra, 2024-04-10:
@ivatet-amd Any news?  Currently there is an old SFC netdriver which does not support linux-6.8 and a new driver which has a completely broken RSS parameters.  Oh, and there is in-kernel SFC driver with its own limitations.

ivatet-amd, 2024-04-15:
Hi @ol-alexandra, @ol-yurijp created an internal ticket in January (SWNETLINUX-5215), but the Netdriver team has not reviewed it yet. Let me page it in today and see what I can do.

ol-alexandra, 2024-04-19:
Спасибо!
Thank you!

ivatet-amd, 2024-04-26:
So the `efx_sfctool_get_rxfh()` fix is now checked in at https://github.com/Xilinx-CNS/onload/commit/88ca3c8d4c14e22d5aeb9d2bba43002c8fef9636. Please reopen if the issue still persists!

closed 2024-04-26 by ivatet-amd