Xilinx-CNS/onload #209: Onload DKMS fails to build on Rocky 9.3 kernel 5.14.0-362.18.1.el9_3.0.1.x86_64

elcabaye, 2024-03-15:
```
# cat /etc/os-release 
NAME="Rocky Linux"
VERSION="9.3 (Blue Onyx)"
ID="rocky"
ID_LIKE="rhel centos fedora"
VERSION_ID="9.3"
PLATFORM_ID="platform:el9"
PRETTY_NAME="Rocky Linux 9.3 (Blue Onyx)"
ANSI_COLOR="0;32"
LOGO="fedora-logo-icon"
CPE_NAME="cpe:/o:rocky:rocky:9::baseos"
HOME_URL="https://rockylinux.org/"
BUG_REPORT_URL="https://bugs.rockylinux.org/"
SUPPORT_END="2032-05-31"
ROCKY_SUPPORT_PRODUCT="Rocky-Linux-9"
ROCKY_SUPPORT_PRODUCT_VERSION="9.3"
REDHAT_SUPPORT_PRODUCT="Rocky Linux"
REDHAT_SUPPORT_PRODUCT_VERSION="9.3"

# uname -r
5.14.0-362.18.1.el9_3.0.1.x86_64

# dnf -y install onload-dkms-8.1.2.26-0.noarch
Install  1 Package

Total download size: 3.4 M
Installed size: 3.4 M
Downloading Packages:
onload-dkms-8.1.2.26-0.noarch.rpm                81 MB/s | 3.4 MB     00:00    
--------------------------------------------------------------------------------
Total                                            77 MB/s | 3.4 MB     00:00     
Running transaction check
Transaction check succeeded.
Running transaction test
Transaction test succeeded.
Running transaction
  Preparing        :                                                        1/1 
  Running scriptlet: onload-dkms-8.1.2.26-0.noarch                          1/1 
  Installing       : onload-dkms-8.1.2.26-0.noarch                          1/1 
  Running scriptlet: onload-dkms-8.1.2.26-0.noarch                          1/1 
Deprecated feature: REMAKE_INITRD (/tmp/dkms.cVINuB/dkms_source_tree/dkms.conf)

Loading tarball for onload-8.1.2.26
Deprecated feature: REMAKE_INITRD (/tmp/dkms.cVINuB/dkms_source_tree/dkms.conf)
Deprecated feature: REMAKE_INITRD (/usr/src/onload-8.1.2.26/dkms.conf)
Creating symlink /var/lib/dkms/onload/8.1.2.26/source -> /usr/src/onload-8.1.2.26
Sign command: /lib/modules/5.14.0-362.18.1.el9_3.0.1.x86_64/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub
Deprecated feature: REMAKE_INITRD (/var/lib/dkms/onload/8.1.2.26/source/dkms.conf)

Building module:
Cleaning build area...(bad exit status: 2)
/var/lib/dkms/onload/8.1.2.26/build/scripts/onload_install...(bad exit status: 1)
Error! Bad return status for module build on kernel: 5.14.0-362.18.1.el9_3.0.1.x86_64 (x86_64)
Consult /var/lib/dkms/onload/8.1.2.26/build/make.log for more information.
Sign command: /lib/modules/5.14.0-362.18.1.el9_3.0.1.x86_64/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub
Deprecated feature: REMAKE_INITRD (/var/lib/dkms/onload/8.1.2.26/source/dkms.conf)

Building module:
Cleaning build area...(bad exit status: 2)
/var/lib/dkms/onload/8.1.2.26/build/scripts/onload_install...(bad exit status: 1)
Error! Bad return status for module build on kernel: 5.14.0-362.18.1.el9_3.0.1.x86_64 (x86_64)
Consult /var/lib/dkms/onload/8.1.2.26/build/make.log for more information.

  Verifying        : onload-dkms-8.1.2.26-0.noarch                          1/1 

Installed:
  onload-dkms-8.1.2.26-0.noarch                                                 

Complete!

# cat /var/lib/dkms/onload/8.1.2.26/build/make.log
DKMS make.log for onload-8.1.2.26 for kernel 5.14.0-362.18.1.el9_3.0.1.x86_64 (x86_64)
Fri Mar 15 12:51:13 PM EDT 2024
onload_install: Building Onload.
Build tree made for linux as x86_64_linux-5.14.0-362.18.1.el9_3.0.1.x86_64
make: Entering directory '/var/lib/dkms/onload/8.1.2.26/build/build/x86_64_linux-5.14.0-362.18.1.el9_3.0.1.x86_64'
/var/lib/dkms/onload/8.1.2.26/build/src/driver/linux_resource/kernel_compat.sh -k /lib/modules/5.14.0-362.18.1.el9_3.0.1.x86_64/build -q > driver/linux_resource/autocompat.h
kernel_compat.sh: Kernel build tree is unable to build modules
make: *** [/var/lib/dkms/onload/8.1.2.26/build/src/mmake.mk:34: driver/linux_resource/autocompat.h] Error 1
make: Leaving directory '/var/lib/dkms/onload/8.1.2.26/build/build/x86_64_linux-5.14.0-362.18.1.el9_3.0.1.x86_64'
onload_build: ERROR: Failed to build driver components.
onload_install: ERROR: Build failed.  Not installing.
```

The generated driver/linux_resource/autocompat.h file is empty.

However, I get a proper autocompat.h if I `cd /var/lib/dkms/onload/8.1.2.26/build/build/x86_64_linux-5.14.0-362.18.1.el9_3.0.1.x86_64` and run `/var/lib/dkms/onload/8.1.2.26/build/src/driver/linux_resource/kernel_compat.sh -k /lib/modules/5.14.0-362.18.1.el9_3.0.1.x86_64/build -q > driver/linux_resource/autocompat.h` by hand.

ivatet-amd, 2024-04-25:
Hi @elcabaye, sorry I haven't seen this issue until today!

Are you still experiencing it? It's evidently some sort of configuration or environment problem.

Would it build if you extract the sources `SF-120887-LS-14-OpenOnload-DKMS-Package.zip`/`onload-dkms-8.1.2.26-0.noarch.rpm`/`onload-8.1.2.26.dkms.tar.gz` and run `scripts/onload_build --kernel` in `dkms_source_tree`? If not, it's worth comparing `env` when running `kernel_compat.sh` from within the build scripts vs. standalone. I suppose things like `export CC` from the build files might have caused it.

elcabaye, 2024-04-25:
Hi @ivatet-amd,

Yes, I found a solution that appears to work for me.  I prefixed the MAKE[0] lines with `unset CC;` and the builds worked.

I was attempting to create a branch to upload the change but was not permitted to.

Thanks so much for following up!

Corey

abower-amd, 2024-04-25:
Hi @elcabaye,

> I was attempting to create a branch to upload the change but was not permitted to.

If you want to raise a pull request you can do it by forking this repo and creating the pull request from your clone to this one.

ivatet-amd, 2024-06-26:
Thank you for your PR https://github.com/Xilinx-CNS/onload/pull/218, which we merged last week.

closed 2024-06-26 by ivatet-amd