Xilinx-CNS/onload #210: Failed to finish install Openonload on AWS EC2

moscovium115, 2024-03-17:
Hello, I'm trying to reduce the latency of an application im working with it, and figured that openonload would be excellent for this, since it does not require me to adapt my application as with DPDK. Unfortunately I encountered some problems. I'm trying to install openonload on AWS EC2 with a c5n.xlarge instance on ubuntu.

I followed these instructions:

```

# install dependencies
sudo apt-get update -y
sudo apt-get install -y --no-install-recommends \
        autoconf \
        automake \
        ca-certificates \
        coreutils \
        curl \
        ethtool \
        g++ \
        gcc \
        kmod \
        linux-libc-dev \
        libpcap0.8-dev \
        libtool-bin \
        make \
        net-tools \
        patch \
        perl \
        python2-dev \
        sed \
        tar \
        unzip \
        util-linux \
        valgrind \
        wget

# Download and extract OpenOnload
export ONLOAD_PACKAGE_URL="https://www.xilinx.com/content/dam/xilinx/publications/solarflare/onload/openonload/8_1_2_26/SF-109585-LS-43-OpenOnload-Package.zip"
curl -fSL $ONLOAD_PACKAGE_URL -o SF-109585-LS-43-OpenOnload-Package.zip
unzip SF-109585-LS-43-OpenOnload-Package.zip
tar xzf onload-8.1.2.26.tgz

# Install OpenOnload
cd onload-8.1.2.26
sudo ./scripts/onload_install

```

upon the last command "sudo ./scripts/onload_install" we see: 
```
onload_install: Install complete.

onload_install: To load the newly installed drivers run:  onload_tool reload
```
however I get the following:

```
root@ip-172-31-0-189:/home/ubuntu/onload-8.1.2.26# onload_tool reload
onload_tool: /sbin/modprobe sfc
modprobe: ERROR: could not insert 'sfc': Unknown symbol in module, or unknown parameter (see dmesg)
modprobe: ERROR: ../libkmod/libkmod-module.c:990 command_do() Error running install command '/sbin/modprobe --ignore-install sfc  &&             if [ -e /lib/modules/$(uname -r)/extra/sfc_aoe.ko ]; then               /sbin/modprobe sfc_aoe;             fi' for module sfc: retcode 1
modprobe: ERROR: could not insert 'sfc': Invalid argument
onload_tool: FAILED: /sbin/modprobe sfc
```
Any advice for getting openonload working on amazon ec2 is much appreciated. 

Kind regards :)

ivatet-amd, 2024-03-18:
Hi @moscovium115, for this particular runtime error mode, please explore the `--no-sfc` option in the `onload_install` script. It should build the Onload without the Netdriver (`sfc.ko`) support, which you do not need in the public cloud.

Alternatively, check `dmesg` for the unresolved symbol messages and try addressing them manually by loading the corresponding module. For instance, if you see `sfc: Unknown symbol mtd_device_unregister`, try `modprobe mtd` and reload the Onload drivers.

Going further, you likely want to use the AF_XDP mode in AWS. If you need help, please email [support-nic@amd.com](mailto:support-nic@amd.com). If this is your first time doing this, you should receive an automated email with the details to sign up.

CC @matthewr-xilinx in the light of the relevant [ON-15479](https://jira.xilinx.com/browse/ON-15479).

moscovium115, 2024-03-18:
Thanks for your reaction @ivatet-amd. 
`sudo ./scripts/onload_install --no-sfc
...
`
```
root@ip-172-31-13-139:/home/ubuntu/onload-8.1.2.26# onload_tool reload
onload_tool: /sbin/modprobe sfc
modprobe: ERROR: ../libkmod/libkmod-module.c:838 kmod_module_insert_module() could not find module by name='sfc'
modprobe: ERROR: could not insert 'sfc': Unknown symbol in module, or unknown parameter (see dmesg)
modprobe: ERROR: ../libkmod/libkmod-module.c:990 command_do() Error running install command '/sbin/modprobe --ignore-install sfc  &&             if [ -e /lib/modules/$(uname -r)/extra/sfc_aoe.ko ]; then               /sbin/modprobe sfc_aoe;             fi' for module sfc: retcode 1
modprobe: ERROR: could not insert 'sfc': Invalid argument
onload_tool: FAILED: /sbin/modprobe sfc
```

I don't see anything relating to openonload in` dmesg`. Any ideas? 
Where can I find @matthewr-xilinx 's email to CC in the email for support for enabling AF_XDP?

abower-amd, 2024-03-18:
> Thanks for your reaction @ivatet-amd. `sudo ./scripts/onload_install --no-sfc`

Hi @moscovium115, thanks for trying out Onload!

> ```
> root@ip-172-31-13-139:/home/ubuntu/onload-8.1.2.26# onload_tool reload
> ```

Please read the three example lines listed at https://github.com/Xilinx-CNS/onload#building-without-xilinx-nics-for-af_xdp-only carefully - it looks to like you may need to vary how you invoke `onload_tool`, not just the installation step.

> email for support for enabling AF_XDP on aws?

Apologies for any confusion on the support status of AF_XDP. Please note that AF_XDP is not a _commercially supported_ use case for Onload. Although you may register any interest in commercial support for Onload on AF_XDP via a sales representative the best place to go for help with using Onload on AF_XDP is here in this github repository for community support.

> ```
> # Download and extract OpenOnload
> export ONLOAD_PACKAGE_URL="https://www.xilinx.com/content/dam/xilinx/publications/solarflare/onload/openonload/8_1_2_26/SF-109585-LS-43-OpenOnload-Package.zip"
> ```

For AF_XDP you would be best advised to use the latest code on github to benefit from enhancements which have not yet featured in the supported releases for AMD Solarflare network adapters. You can use [`onload_mkdist`](https://github.com/Xilinx-CNS/onload/blob/master/DEVELOPING.md#distributing-as-tarball) to generate a release-like package which has the benefit of generating a suitable version number to reflect the status of the source.

moscovium115, 2024-03-18:
Hi @abower-amd, much thanks for writing a thorough response.

Thank you!` onload_tool reload --onload-only` fixed the error. I want to make sure I installed from the repository right:

```
#install dependencies
sudo apt-get update -y
sudo apt-get install -y --no-install-recommends \
        autoconf \
        automake \
        ca-certificates \
        coreutils \
        curl \
        ethtool \
        g++ \
        gcc \
        kmod \
        linux-libc-dev \
        libpcap0.8-dev \
        libtool-bin \
        make \
        net-tools \
        patch \
        perl \
        python2-dev \
        sed \
        tar \
        unzip \
        util-linux \
        valgrind \
        wget

sudo apt-get install libcap-dev

# Download and extract OpenOnload
git clone https://github.com/Xilinx-CNS/onload.git 
cd  onload/
scripts/onload_mkdist
tar xzf onload-8010666.tgz
cd onload-8010666

#install Openonload
sudo ./scripts/onload_install --no-sfc
sudo onload_tool reload --onload-only
```
When using a rust script to measure my latency:
`onload cargo run --release`

after the rust compilation I see this warning:
```
oo:Ping_bybit[20077]: netif_tcp_helper_alloc_u: ENODEV.
This error can occur if:
 - no Solarflare network interfaces are active/UP, or they are running
   packed stream firmware or are disabled, and
 - there are no AF_XDP interfaces registered with sfc_resource
Please check your configuration.
```
Consequently, I do not see any speed up from using openonload. Could this be caused by the warning? How to resolve the warnings?

ivatet-amd, 2024-03-19:
Hi @moscovium115, I guess that your EC2 kernel panics and freezes up as a result. Do you reckon this is the case?

Double-checking: Have you tried vice versa: setting a smaller MTU and then registering the interface with Onload?

I suspect a bug in [efrm_nic_add()](https://github.com/Xilinx-CNS/onload/blob/801066643e8d306027e6177eeea0e75b8ac33540/src/driver/linux_resource/resource_driver.c#L439). It should take the `failed` path on error but instead it goes ahead and adds an unitialised `efrm_nic` into Onload, and presumably causes the kernel panic.

moscovium115, 2024-03-19:
Hi @ivatet-amd, 
I do think the kernel panics indeed!

```
root@ip-172-31-6-241:/home/ubuntu# dmesg | grep -i panic
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-6.5.0-1014-aws root=PARTUUID=b4b91ea5-ac2a-4289-9a10-c53cba170810 ro console=tty1 console=ttyS0 nvme_core.io_timeout=4294967295 panic=-1
[    0.026187] Kernel command line: BOOT_IMAGE=/boot/vmlinuz-6.5.0-1014-aws root=PARTUUID=b4b91ea5-ac2a-4289-9a10-c53cba170810 ro console=tty1 console=ttyS0 nvme_core.io_timeout=4294967295 panic=-1
```
here it panicked twice since I tried two different ways to register the interface.
if i first set a smaller mtu and then register interface with Onload and then look at dmesg i see:
```
[  139.292831] [sfc efrm] efrm_nondl_register_device: register ens5
[  139.293550] [sfc efrm] ens5 type=4:X0 ifindex=2
[  139.293737] ena 0000:00:05.0 ens5: Failed to set xdp program, the Rx/Tx channel count should be at most half of the maximum allowed channel count. The current queue count (4), the maximal queue count (4)
[  139.295754] [sfc efrm] ?: ERROR: hardware init failed rc=-22
[  139.296418] [sfc efrm] ens5 index=0 ifindex=2
[  139.296430] [onload] oo_nic_add: ifindex=2 oo_index=0
```
using onload ping google.com the system freezes again.

ivatet-amd, 2024-03-19:
Thanks, @moscovium115, for the prompt response. It looks like the `ens5` network driver refuses to link the XDP program and fails later on.

Could you try this patch to see if it helps to resolve the kernel panic:
```diff
diff --git a/src/driver/linux_resource/resource_driver.c b/src/driver/linux_resource/resource_driver.c
index 7193a18ccb..063089a42b 100644
--- a/src/driver/linux_resource/resource_driver.c
+++ b/src/driver/linux_resource/resource_driver.c
@@ -440,6 +440,7 @@ efrm_nic_add(void *drv_device, struct device *dev,
                /* Again, PCI VFs may be available. */
                EFRM_ERR("%s: ERROR: hardware init failed rc=%d",
                         dev && dev_name(dev) ? dev_name(dev) : "?", rc);
+               goto failed;
        }
        efrm_resource_manager_add_total(EFRM_RESOURCE_VI,
                                        efrm_nic->max_vis);
```

Could you also try this patch to try the generic XDP mode (in contrast to the native XDP mode that the EC network driver refuses):
```diff
diff --git a/src/lib/efhw/af_xdp.c b/src/lib/efhw/af_xdp.c
index 2a48bb0b5c..8a579c7eb7 100644
--- a/src/lib/efhw/af_xdp.c
+++ b/src/lib/efhw/af_xdp.c
@@ -478,7 +478,7 @@ static int xdp_bind(struct socket* sock, int ifindex, unsigned queue, unsigned f
 /* Link an XDP program to an interface */
 static int xdp_set_link(struct net_device* dev, int prog_fd)
 {
-  if( dev->netdev_ops->ndo_bpf ) {
+  if( 0 ) {
     struct netdev_bpf bpf = {
       .command = XDP_SETUP_PROG,
       .prog = NULL,
```
Note: Please make sure the Onload kernel "knows" about the helper utility `bpf-link-helper` ( https://github.com/Xilinx-CNS/onload?tab=readme-ov-file#compatible-network-adapters-drivers-and-operating-systems).

moscovium115, 2024-03-19:
Thank you @ivatet-amd,

regarding the helper utility:
```
root@ip-172-31-6-241:/home/ubuntu/onload# make -C ./src/tools/bpf_link_helper/
make: Entering directory '/home/ubuntu/onload/src/tools/bpf_link_helper'
cc  bpf-link-helper.c -o bpf-link-helper 
make: Leaving directory '/home/ubuntu/onload/src/tools/bpf_link_helper'
root@ip-172-31-6-241:/home/ubuntu/onload# echo $(realpath ./src/tools/bpf_link_helper/bpf-link-helper) | sudo tee /sys/module/sfc_resource/parameters/bpf_link_helper
/home/ubuntu/onload/src/tools/bpf_link_helper/bpf-link-helper

Excuse my ignorance for I'm not used to patching files, but is the correct procedure as follows?
```
nano patch_1.patch
#add patch in .patch file 
git apply patch_1.patch
```

This failed for the first patch to resolve kernel panic:
```
root@ip-172-31-6-241:/home/ubuntu/onload# git apply patch_one.patch 
error: patch failed: src/driver/linux_resource/resource_driver.c:440
error: src/driver/linux_resource/resource_driver.c: patch does not apply
```
It did work however for the second patch you gave for generic xdp. I just manually added the line: `goto failed;` to get the first patch working if thats alright?

is this how  src/driver/linux_resource/resource_driver.c should look like after patch?
```
439         if (rc < 0) {
440                 /* Again, PCI VFs may be available. */
441                 EFRM_ERR("%s: ERROR: hardware init failed rc=%d",
442                          dev && dev_name(dev) ? dev_name(dev) : "?", rc);
443                 goto failed;
444         }
445         efrm_resource_manager_add_total(EFRM_RESOURCE_VI,
446                                         efrm_nic->max_vis);
447         efrm_resource_manager_add_total(EFRM_RESOURCE_PD,
448                                         efrm_nic->max_vis);
449
450         /* Tell NIC to spread wakeup events. */
```

Stupid question:  I presume these patches should be applied before the installation(sudo ./scripts/onload_install --no-sfc)?
After applying patches, I install, change mtu and add interface to afxdp:
```

root@ip-172-31-6-241:/home/ubuntu/onload# sudo ./scripts/onload_install --no-sfc
root@ip-172-31-6-241:/home/ubuntu/onload# sudo ip link set dev ens5 mtu 1500
root@ip-172-31-6-241:/home/ubuntu/onload# echo ens5 > /sys/module/sfc_resource/afxdp/register
```

sadly dmesg looks like:
```
[  531.929792] [sfc efrm] efrm_nondl_register_device: register ens5
[  531.930511] [sfc efrm] ens5 type=4:X0 ifindex=2
[  531.930686] ena 0000:00:05.0 ens5: Failed to set xdp program, the Rx/Tx channel count should be at most half of the maximum allowed channel count. The current queue count (4), the maximal queue count (4)
[  531.932750] [sfc efrm] ?: ERROR: hardware init failed rc=-22
[  531.933500] [sfc efrm] ens5 index=0 ifindex=2
[  531.933511] [onload] oo_nic_add: ifindex=2 oo_index=0
```
now using `onload ping google.com`:

```
root@ip-172-31-6-241:/home/ubuntu/onload# onload ping google.com
citp_oo_get_cpu_khz: Failed to open /dev/onload
oo_onloadfs_dev_t: Failed to open /dev/onload
oo_onloadfs_dev_t: Failed to open /dev/onload
PING google.com (142.251.175.101) 56(84) bytes of data.
oo_onloadfs_dev_t: Failed to open /dev/onload
64 bytes from sh-in-f101.1e100.net (142.251.175.101): icmp_seq=1 ttl=56 time=1.47 ms
64 bytes from sh-in-f101.1e100.net (142.251.175.101): icmp_seq=2 ttl=56 time=1.50 ms
64 bytes from sh-in-f101.1e100.net (142.251.175.101): icmp_seq=3 ttl=56 time=1.49 ms

```
we get  different errors, but the kernel doesn't panic anymore which means your patch worked. Sadly there is no speed up yet.

abower-amd, 2024-03-19:
Hi @moscovium115,

This is not a complete answer but a couple of things to note:

> ```
> citp_oo_get_cpu_khz: Failed to open /dev/onload
> oo_onloadfs_dev_t: Failed to open /dev/onload
> oo_onloadfs_dev_t: Failed to open /dev/onload
> oo_onloadfs_dev_t: Failed to open /dev/onload
> ```

This means that the `onload` kernel module has failed to load - if this does not succeed then nothing will ever get accelerated by onload, so best to solve this first before trying to work with traffic.

> now using onload ping google.com:

Onload does not accelerate ICMP so you will never see a `ping` speed-up. You need to use a UDP (e.g. [sfnt-pingpong](https://github.com/Xilinx-CNS/cns-sfnettest)) or TCP (e.g. _netcat_) application to test Onload.

moscovium115, 2024-03-19:
@abower-amd Thank you for the informative explanation! I should have payed more attention to what `ping `actually does.
My application that I want to latency reduce  does use TCP via Rust's hyper library.

moscovium115, 2024-03-20:
Apologies @ivatet-amd @abower-amd My second to last comment contained some mistakes, which I have now fixed.

moscovium115, 2024-03-22:
Hello, wanted to give some information. I was able to install without the no sfc option by installing `linux-modules-extra`.
My install proces now looks like:
`sudo apt install libcap-dev
sudo apt install build-essential
git clone https://github.com/Xilinx-CNS/onload.git && cd onload
sudo scripts/onload_install
sudo apt install linux-modules-extra-$(uname -r)
sudo onload_tool reload
sudo ip link set dev ens5  mtu 3498
sudo ethtool -L ens5 combined 1
echo ens5 | sudo tee /sys/module/sfc_resource/afxdp/register`

From related github issues (https://github.com/Xilinx-CNS/onload/issues/78, https://github.com/amzn/amzn-drivers/issues/241) I understand the combined channels must be set to 1. 

Using this I can see in syslog:
[image]
Upon using `$ sudo onload iperf -c  172.31.2.94 -M 64 -t 20 -i 1` I noticed that throughput is actually reduced using openonload:
openonload:

```
ubuntu@ip-172-31-9-80:~/onload$ sudo onload iperf -c  172.31.2.94 -M 64 -t 20 -i 1
oo:iperf[33051]: Using Onload 13f04401 2024-03-21 master  [3]
oo:iperf[33051]: Copyright (c) 2002-2024 Advanced Micro Devices, Inc.
WARNING: attempt to set TCP maxmimum segment size to 64 failed.
Setting the MSS may not be implemented on this OS.
------------------------------------------------------------
Client connecting to 172.31.2.94, TCP port 5001
TCP window size:  128 KByte (default)
------------------------------------------------------------
[  1] local 172.31.9.80 port 33125 connected with 172.31.2.94 port 5001 (MSS=1700)
[ ID] Interval       Transfer     Bandwidth
[  1] 0.0000-1.0000 sec   815 MBytes  6.83 Gbits/sec
[  1] 1.0000-2.0000 sec   836 MBytes  7.01 Gbits/sec
[  1] 2.0000-3.0000 sec   828 MBytes  6.94 Gbits/sec
[  1] 3.0000-4.0000 sec   831 MBytes  6.97 Gbits/sec
[  1] 4.0000-5.0000 sec   848 MBytes  7.11 Gbits/sec
[  1] 5.0000-6.0000 sec   826 MBytes  6.93 Gbits/sec
[  1] 6.0000-7.0000 sec   828 MBytes  6.95 Gbits/sec
[  1] 7.0000-8.0000 sec   820 MBytes  6.88 Gbits/sec
[  1] 8.0000-9.0000 sec   859 MBytes  7.20 Gbits/sec
[  1] 9.0000-10.0000 sec   864 MBytes  7.25 Gbits/sec
[  1] 10.0000-11.0000 sec   862 MBytes  7.23 Gbits/sec
[  1] 11.0000-12.0000 sec   868 MBytes  7.28 Gbits/sec
[  1] 12.0000-13.0000 sec   877 MBytes  7.35 Gbits/sec
[  1] 13.0000-14.0000 sec   877 MBytes  7.36 Gbits/sec
[  1] 14.0000-15.0000 sec   864 MBytes  7.25 Gbits/sec
[  1] 15.0000-16.0000 sec   861 MBytes  7.22 Gbits/sec
[  1] 16.0000-17.0000 sec   844 MBytes  7.08 Gbits/sec
[  1] 17.0000-18.0000 sec   836 MBytes  7.01 Gbits/sec
[  1] 18.0000-19.0000 sec   858 MBytes  7.20 Gbits/sec
[  1] 19.0000-20.0000 sec   845 MBytes  7.09 Gbits/sec
[  1] 0.0000-20.0024 sec  16.5 GBytes  7.11 Gbits/sec
```

no openonload:

```
ubuntu@ip-172-31-9-80:~/onload$ sudo iperf -c  172.31.2.94 -M 64 -t 20 -i 1
WARNING: attempt to set TCP maxmimum segment size to 64 failed.
Setting the MSS may not be implemented on this OS.
------------------------------------------------------------
Client connecting to 172.31.2.94, TCP port 5001
TCP window size:  325 KByte (default)
------------------------------------------------------------
[  1] local 172.31.9.80 port 50876 connected with 172.31.2.94 port 5001 (MSS=3446)
[ ID] Interval       Transfer     Bandwidth
[  1] 0.0000-1.0000 sec  1.10 GBytes  9.45 Gbits/sec
[  1] 1.0000-2.0000 sec  1.10 GBytes  9.42 Gbits/sec
[  1] 2.0000-3.0000 sec  1.10 GBytes  9.42 Gbits/sec
[  1] 3.0000-4.0000 sec  1.10 GBytes  9.42 Gbits/sec
[  1] 4.0000-5.0000 sec  1.10 GBytes  9.42 Gbits/sec
[  1] 5.0000-6.0000 sec  1.10 GBytes  9.42 Gbits/sec
[  1] 6.0000-7.0000 sec  1.10 GBytes  9.42 Gbits/sec
[  1] 7.0000-8.0000 sec  1.10 GBytes  9.42 Gbits/sec
[  1] 8.0000-9.0000 sec  1.10 GBytes  9.41 Gbits/sec
[  1] 9.0000-10.0000 sec  1.10 GBytes  9.43 Gbits/sec
[  1] 10.0000-11.0000 sec  1.10 GBytes  9.42 Gbits/sec
[  1] 11.0000-12.0000 sec  1.10 GBytes  9.42 Gbits/sec
[  1] 12.0000-13.0000 sec  1.10 GBytes  9.42 Gbits/sec
[  1] 13.0000-14.0000 sec  1.10 GBytes  9.42 Gbits/sec
[  1] 14.0000-15.0000 sec  1.10 GBytes  9.43 Gbits/sec
[  1] 15.0000-16.0000 sec  1.10 GBytes  9.42 Gbits/sec
[  1] 16.0000-17.0000 sec  1.10 GBytes  9.43 Gbits/sec
[  1] 17.0000-18.0000 sec  1.10 GBytes  9.42 Gbits/sec
[  1] 18.0000-19.0000 sec  1.10 GBytes  9.42 Gbits/sec
[  1] 19.0000-20.0000 sec  1.10 GBytes  9.42 Gbits/sec
[  1] 0.0000-20.0133 sec  21.9 GBytes  9.42 Gbits/sec

```

dmesg looks as follows:
```
[ 2039.566084] [onload] [3]: WARNING: huge pages are incompatible with AF_XDP. Disabling hugepage support.
[ 2039.610737] ena 0000:00:05.0 ens5: Command parameter 46 is not supported
[ 2039.611975] [sfc efhw] rxclass_get_dev_info: rxclass: Cannot get RX class rule count
[ 2039.611980] [onload] oof_local_port_addr_fixup_wild: 3:2047 ERROR: FILTER TCP 172.31.9.80:33125 0.0.0.0:0 failed (-95)
[ 2109.625388] [sfc efhw] af_xdp_flush_rx_dma_channel: FIXME AF_XDP
[ 2109.626513] [sfc efhw] af_xdp_flush_tx_dma_channel: FIXME AF_XDP

```

abower-amd, 2024-03-22:
> I noticed that throughput is actually reduced.

I thought your application was targeting latency? I suggest reading up on [Onload tuning](https://docs.amd.com/r/en-US/ug1586-onload-user/Using-Onload-Tuning-Profiles). Note that not everything will directly apply to AF_XDP and that whether the underlying network interface driver supports zero copy will have an impact on what can be achieved.

abower-amd, 2024-04-16:
Closing this issue as the original problem is now resolved by advice. Please raise another issue if you'd like to discuss the performance question further! Thanks for trying out Onload for your application.

closed 2024-04-16 by abower-amd

beanqi, 2025-03-11:
> Hello, wanted to give some information. I was able to install without the no sfc option by installing `linux-modules-extra`. My install proces now looks like:
> 
> ```
> sudo apt install libcap-dev
> sudo apt install build-essential
> git clone https://github.com/Xilinx-CNS/onload.git && cd onload
> sudo scripts/onload_install
> sudo apt install linux-modules-extra-$(uname -r)
> sudo onload_tool reload
> sudo ip link set dev ens5  mtu 3498
> sudo ethtool -L ens5 combined 1
> echo ens5 | sudo tee /sys/module/sfc_resource/afxdp/register
> ```
> 
> From related github issues ([#78](https://github.com/Xilinx-CNS/onload/issues/78), [amzn/amzn-drivers#241](https://github.com/amzn/amzn-drivers/issues/241)) I understand the combined channels must be set to 1.
> 
> Using this I can see in syslog: [image] Upon using `$ sudo onload iperf -c 172.31.2.94 -M 64 -t 20 -i 1` I noticed that throughput is actually reduced.
> 
> openonload:
> 
> ```
> ubuntu@ip-172-31-9-80:~/onload$ sudo onload iperf -c  172.31.2.94 -M 64 -t 20 -i 1
> oo:iperf[33051]: Using Onload 13f04401 2024-03-21 master  [3]
> oo:iperf[33051]: Copyright (c) 2002-2024 Advanced Micro Devices, Inc.
> WARNING: attempt to set TCP maxmimum segment size to 64 failed.
> Setting the MSS may not be implemented on this OS.
> ------------------------------------------------------------
> Client connecting to 172.31.2.94, TCP port 5001
> TCP window size:  128 KByte (default)
> ------------------------------------------------------------
> [  1] local 172.31.9.80 port 33125 connected with 172.31.2.94 port 5001 (MSS=1700)
> [ ID] Interval       Transfer     Bandwidth
> [  1] 0.0000-1.0000 sec   815 MBytes  6.83 Gbits/sec
> [  1] 1.0000-2.0000 sec   836 MBytes  7.01 Gbits/sec
> [  1] 2.0000-3.0000 sec   828 MBytes  6.94 Gbits/sec
> [  1] 3.0000-4.0000 sec   831 MBytes  6.97 Gbits/sec
> [  1] 4.0000-5.0000 sec   848 MBytes  7.11 Gbits/sec
> [  1] 5.0000-6.0000 sec   826 MBytes  6.93 Gbits/sec
> [  1] 6.0000-7.0000 sec   828 MBytes  6.95 Gbits/sec
> [  1] 7.0000-8.0000 sec   820 MBytes  6.88 Gbits/sec
> [  1] 8.0000-9.0000 sec   859 MBytes  7.20 Gbits/sec
> [  1] 9.0000-10.0000 sec   864 MBytes  7.25 Gbits/sec
> [  1] 10.0000-11.0000 sec   862 MBytes  7.23 Gbits/sec
> [  1] 11.0000-12.0000 sec   868 MBytes  7.28 Gbits/sec
> [  1] 12.0000-13.0000 sec   877 MBytes  7.35 Gbits/sec
> [  1] 13.0000-14.0000 sec   877 MBytes  7.36 Gbits/sec
> [  1] 14.0000-15.0000 sec   864 MBytes  7.25 Gbits/sec
> [  1] 15.0000-16.0000 sec   861 MBytes  7.22 Gbits/sec
> [  1] 16.0000-17.0000 sec   844 MBytes  7.08 Gbits/sec
> [  1] 17.0000-18.0000 sec   836 MBytes  7.01 Gbits/sec
> [  1] 18.0000-19.0000 sec   858 MBytes  7.20 Gbits/sec
> [  1] 19.0000-20.0000 sec   845 MBytes  7.09 Gbits/sec
> [  1] 0.0000-20.0024 sec  16.5 GBytes  7.11 Gbits/sec
> ```
> 
> no openonload:
> 
> ```
> ubuntu@ip-172-31-9-80:~/onload$ sudo iperf -c  172.31.2.94 -M 64 -t 20 -i 1
> WARNING: attempt to set TCP maxmimum segment size to 64 failed.
> Setting the MSS may not be implemented on this OS.
> ------------------------------------------------------------
> Client connecting to 172.31.2.94, TCP port 5001
> TCP window size:  325 KByte (default)
> ------------------------------------------------------------
> [  1] local 172.31.9.80 port 50876 connected with 172.31.2.94 port 5001 (MSS=3446)
> [ ID] Interval       Transfer     Bandwidth
> [  1] 0.0000-1.0000 sec  1.10 GBytes  9.45 Gbits/sec
> [  1] 1.0000-2.0000 sec  1.10 GBytes  9.42 Gbits/sec
> [  1] 2.0000-3.0000 sec  1.10 GBytes  9.42 Gbits/sec
> [  1] 3.0000-4.0000 sec  1.10 GBytes  9.42 Gbits/sec
> [  1] 4.0000-5.0000 sec  1.10 GBytes  9.42 Gbits/sec
> [  1] 5.0000-6.0000 sec  1.10 GBytes  9.42 Gbits/sec
> [  1] 6.0000-7.0000 sec  1.10 GBytes  9.42 Gbits/sec
> [  1] 7.0000-8.0000 sec  1.10 GBytes  9.42 Gbits/sec
> [  1] 8.0000-9.0000 sec  1.10 GBytes  9.41 Gbits/sec
> [  1] 9.0000-10.0000 sec  1.10 GBytes  9.43 Gbits/sec
> [  1] 10.0000-11.0000 sec  1.10 GBytes  9.42 Gbits/sec
> [  1] 11.0000-12.0000 sec  1.10 GBytes  9.42 Gbits/sec
> [  1] 12.0000-13.0000 sec  1.10 GBytes  9.42 Gbits/sec
> [  1] 13.0000-14.0000 sec  1.10 GBytes  9.42 Gbits/sec
> [  1] 14.0000-15.0000 sec  1.10 GBytes  9.43 Gbits/sec
> [  1] 15.0000-16.0000 sec  1.10 GBytes  9.42 Gbits/sec
> [  1] 16.0000-17.0000 sec  1.10 GBytes  9.43 Gbits/sec
> [  1] 17.0000-18.0000 sec  1.10 GBytes  9.42 Gbits/sec
> [  1] 18.0000-19.0000 sec  1.10 GBytes  9.42 Gbits/sec
> [  1] 19.0000-20.0000 sec  1.10 GBytes  9.42 Gbits/sec
> [  1] 0.0000-20.0133 sec  21.9 GBytes  9.42 Gbits/sec
> ```
> 
> dmesg looks as follows:
> 
> ```
> [ 2039.566084] [onload] [3]: WARNING: huge pages are incompatible with AF_XDP. Disabling hugepage support.
> [ 2039.610737] ena 0000:00:05.0 ens5: Command parameter 46 is not supported
> [ 2039.611975] [sfc efhw] rxclass_get_dev_info: rxclass: Cannot get RX class rule count
> [ 2039.611980] [onload] oof_local_port_addr_fixup_wild: 3:2047 ERROR: FILTER TCP 172.31.9.80:33125 0.0.0.0:0 failed (-95)
> [ 2109.625388] [sfc efhw] af_xdp_flush_rx_dma_channel: FIXME AF_XDP
> [ 2109.626513] [sfc efhw] af_xdp_flush_tx_dma_channel: FIXME AF_XDP
> ```

I followed the same instructions, and everything seemed fine. However, when I used the `sudo onload ./xxx` command, SSH went offline, and it appeared as if the server rebooted. Have you encountered this issue