Xilinx-CNS/onload #212: aarch64 build is broken with newdriver build

okt-sergeyn, 2024-03-25:
Onload master 13f04401
Building modules on raspberry pi with 5.15.83 kernel:
```
HAVE_EF10CT=0 HAVE_EFCT=0 HAVE_SFC=0 mmakebuildtree --driver
HAVE_EF10CT=0 HAVE_EFCT=0 HAVE_SFC=0 make -C build/aarch64_linux-5.15.83-v8/
```
Errors:
```
MODPOST /home/pi/onload/build/aarch64_linux-5.15.83-v8/Module.symvers                                                                                                                                         
ERROR: modpost: "__aarch64_ldset4_acq_rel" [/home/pi/onload/build/aarch64_linux-5.15.83-v8/src/driver/linux_resource/sfc_resource.ko] undefined!                                                                
ERROR: modpost: "__aarch64_cas4_acq_rel" [/home/pi/onload/build/aarch64_linux-5.15.83-v8/src/driver/linux_resource/sfc_resource.ko] undefined!                                                                  
ERROR: modpost: "__aarch64_ldclr4_acq_rel" [/home/pi/onload/build/aarch64_linux-5.15.83-v8/src/driver/linux_onload/onload.ko] undefined!                                                                        
ERROR: modpost: "__aarch64_ldadd4_acq_rel" [/home/pi/onload/build/aarch64_linux-5.15.83-v8/src/driver/linux_onload/onload.ko] undefined!                                                                        
ERROR: modpost: "__aarch64_ldset4_acq_rel" [/home/pi/onload/build/aarch64_linux-5.15.83-v8/src/driver/linux_onload/onload.ko] undefined!                                                                        
ERROR: modpost: "__aarch64_swp8_acq" [/home/pi/onload/build/aarch64_linux-5.15.83-v8/src/driver/linux_onload/onload.ko] undefined!                                                                              
ERROR: modpost: "__aarch64_cas8_acq_rel" [/home/pi/onload/build/aarch64_linux-5.15.83-v8/src/driver/linux_onload/onload.ko] undefined!                                                                          
ERROR: modpost: "__aarch64_cas4_acq_rel" [/home/pi/onload/build/aarch64_linux-5.15.83-v8/src/driver/linux_onload/onload.ko] undefined!                                                                          
ERROR: modpost: "__aarch64_swp4_acq" [/home/pi/onload/build/aarch64_linux-5.15.83-v8/src/driver/linux_onload/onload.ko] undefined!                                                                              
ERROR: modpost: "__aarch64_cas8_acq_rel" [/home/pi/onload/build/aarch64_linux-5.15.83-v8/src/driver/linux_char/sfc_char.ko] undefined!                                                                          
WARNING: modpost: suppressed 1 unresolved symbol warnings because there were too many)                                                                                                                             
make[3]: *** [scripts/Makefile.modpost:133: /home/pi/onload/build/aarch64_linux-5.15.83-v8/Module.symvers] Error 1                                                                                              
make[3]: *** Deleting file '/home/pi/onload/build/aarch64_linux-5.15.83-v8/Module.symvers'                                                                                                                      
make[2]: *** [Makefile:1817: modules] Error 2                                                                                                                                                                      
make[2]: Leaving directory '/home/pi/linux-5.15'                                                                                                                                                                   
make[1]: *** [Makefile:254: modules] Error 2                                                                                                                                                                       
make[1]: Leaving directory '/home/pi/onload'                                                                                                                                                                       
make: *** [GNUmakefile:2: all] Error 2 
```
It used to work with old build system.

ivatet-amd, 2024-03-25:
Thanks, @okt-sergeyn. Do you have insights on how long aarch64 has been broken? I probably won't look at it until next week.

okt-sergeyn, 2024-03-25:
It has broken since this commit:
```
commit 3e786841932c7f1f5eca1f974e90ca9f3bf166d1
Author: Siân James <sian.james@amd.com>
Date:   Fri Mar 1 15:08:45 2024 +0000

    ON-15424: Enable newdriver build by default
```
At least, reverting it helped me.

okt-sergeyn, 2024-04-18:
`-mno-outline-atomics` must be passed to gcc. I'm going to publish a patch.

closed 2024-04-22 by ivatet-amd