Xilinx-CNS/onload #213: Possible deadlock while connecting TCP socket

injecto, 2024-03-26:
Hello, I've discovered a thread stuck while socket `connect`:

```
Thread 29 (Thread 0x7fa1c8ff9700 (LWP 441693)):
#0  futex_wait_cancelable (private=<optimized out>, expected=0, futex_word=0x7fa1c8fe6da0) at ../sysdeps/nptl/futex-internal.h:183
#1  __pthread_cond_wait_common (abstime=0x0, clockid=0, mutex=0x7fa212a1a528 <citp_ul_lock+104>, cond=0x7fa1c8fe6d78) at pthread_cond_wait.c:508
#2  __pthread_cond_wait (cond=cond@entry=0x7fa1c8fe6d78, mutex=0x7fa212a1a528 <citp_ul_lock+104>) at pthread_cond_wait.c:647
#3  0x00007fa2129747a2 in oo_rwlock_cond_wait (rwlock=<optimized out>, cond=0x7fa1c8fe6d78) at /root/openonload/onload-8.1.2.26/src/include/onload/ul/rwlock.h:332
#4  citp_fdtable_busy_wait (fd=fd@entry=362, fdt_locked=fdt_locked@entry=0) at /root/openonload/onload-8.1.2.26/src/lib/transport/unix/fdtable.c:1719
#5  0x00007fa212978f8a in citp_fdtable_lookup (fd=fd@entry=362) at /root/openonload/onload-8.1.2.26/src/lib/transport/unix/fdtable.c:1053
#6  0x00007fa21297ae41 in citp_tcp_connect (fdinfo=0x7fa18c0017d0, sa=0x7fa1c8fe6f80, sa_len=16, lib_context=0x7fa1c8fe6ec0) at /root/openonload/onload-8.1.2.26/src/lib/transport/unix/tcp_fd.c:1152
#7  0x00007fa21296433b in onload_connect (fd=362, sa=0x7fa1c8fe6f80, sa_len=16) at /root/openonload/onload-8.1.2.26/src/lib/transport/unix/sockcall_intercept.c:559
#8  0x00005578a7187fef in socket2::sys::connect (fd=-922849888, addr=0x80) at src/sys/unix.rs:814
#9  socket2::socket::Socket::connect (self=<optimized out>, address=0x80) at src/socket.rs:204
...
```

I've attached an [output](https://github.com/Xilinx-CNS/onload/files/14758699/lots.txt) of `onload_stackdump lots`. Could you please provide some reasons that may influence such a behaviour? Thank you.

injecto, 2024-03-26:
I've checked dmesg also:

```
kernel: [onload] __oof_socket_add: 1:1999 ERROR: laddr=10.236.190.129 not local
```

The log is related to bind IP (I'm using dummy interface) of the socket.

injecto, 2024-03-26:
Looks like https://docs.amd.com/r/en-US/ug1586-onload-user/Running-Out-of-Packet-Buffers. Will investigate though.

injecto, 2024-03-28:
Yes, that was about buffers. The issue might be closed, but just to feed my curiosity: why there is the particular behaviour? I suppose it's counterintuitive a bit: it's better to drop something (data, sockets, etc.) than stuck. Could you please explain the details regarding to it?

ivatet-amd, 2024-03-28:
Hi @injecto, thanks for taking a look!

I consulted with my colleagues and want to share our observations:

1. Could it be that the backtrace you posted is a secondary manifestation of running out of the packet buffers, i.e. there is another thread is being blocked while holding a lock, causing other threads to get stuck?
2. There is a configuration knob `EF_TCP_SEND_NONBLOCK_NO_PACKETS_MODE` when set to 1, it would cause intercepted syscalls to return `-ENOBUFS` instead of blocking, which might help your application?
3. Please explore another one, `EF_FDS_MT_SAFE`, that allows stricter concurrency control when set to 0. This would prevent the fdtable corruption in certain situations.
4. The `__oof_socket_add()` is not immediately related to a deadlock, but it means that Onload will not accelerate an interface with this IP address. In your [lots.txt](https://github.com/Xilinx-CNS/onload/files/14758699/lots.txt), there is plenty of "10.236.190.129" being "Onloaded", so it must be OK.
5. I want to note that the newer Onload versions include a bugfix (https://github.com/Xilinx-CNS/onload/commit/bd96256cced8af7dc380ef14a23aaff00ddfab26) that might affect your configuration: please make sure the IP addresses you give to the acceleratable interfaces are non-localhost, i.e. with a scope larger than the "host".

Hope that all makes sense.

injecto, 2024-03-29:
Hi @ivatet-amd, thanks for the detailed answer, I'll investigate the suggested.

ivatet-amd, 2024-04-15:
Thanks @injecto, closing the issue for now. 👍

closed 2024-04-15 by ivatet-amd