Xilinx-CNS/onload #217: Onload fails to build on Linux version 4.19.90-24.4.v2101.ky10.aarch64

FrankYee2024, 2024-04-25:
【environmental message】
[root@localhost scripts]# cat /proc/version
Linux version 4.19.90-24.4.v2101.ky10.aarch64 (KYLINSOFT@localhost.localdomain) (gcc version 7.3.0 (GCC)) #1 SMP Mon May 24 14:45:37 CST 2021
[root@localhost scripts]# uname -a
Linux localhost.localdomain 4.19.90-24.4.v2101.ky10.aarch64 #1 SMP Mon May 24 14:45:37 CST 2021 aarch64 aarch64 aarch64 GNU/Linux
### Detail steps
1、download Onload source code from:
https://github.com/Xilinx-CNS/onload
2、upload source code to host Server , unzip and build it.
[root@localhost yzm]# cd onload-master/
[root@localhost onload-master]# ls
CONTRIBUTING.md  Gemfile     Jenkinsfile  Makefile  Rakefile  README.md  src
DEVELOPING.md    imports.mk  LICENSE      mk        rakelib   scripts    versions.yaml
[root@localhost onload-master]# cd scripts/
[root@localhost scripts]# ls
anonymise_stackdump.py  jenkins                mmake           onload_install      onload_mktest           onload_version_gen  shell-fns
check_header_presence   lib                    mmakebuildtree  onload_iptables     onload_profiles         run_unit_tests.sh   sockbatch
check_library_presence  libc_compat.sh         mmaketool       onload_misc         onload_rpmtest          sfcaffinity_config
check_prototype         license-blacklist.txt  onload          onload_mkcontainer  onload_tcpdump          sfcaffinity.py
ContainerFileNotice     license-validate.py    onload_apps     onload_mkdist       onload_tool             sfcirqaffinity
debian                  makelinuxinstall       onload_build    onload_mkdkms       onload-user.Dockerfile  sfcmask.py
[root@localhost scripts]# ./onload_install
onload_install: Building Onload.

3、error massages:

/onload_install
onload_install: Building Onload.
Build tree made for linux as aarch64_linux-4.19.90-24.4.v2101.ky10.aarch64
make: 进入目录“/usr/yzm/onload-master/build/aarch64_linux-4.19.90-24.4.v2101.ky10.aarch64”
make -C ../.. KPATH=/lib/modules/4.19.90-24.4.v2101.ky10.aarch64/build KBUILDTOP=/usr/yzm/onload-master/build/aarch64_linux-4.19.90-24.4.v2101.ky10.aarch64 kernel
make[1]: 进入目录“/usr/yzm/onload-master”
make[2]: 进入目录“/usr/src/kernels/4.19.90-24.4.v2101.ky10.aarch64”
(/usr/yzm/onload-master/src/driver/linux_resource/kernel_compat.sh -k /usr/src/kernels/4.19.90-24.4.v2101.ky10.aarch64 -q > /usr/yzm/onload-master/build/aarch64_linux-4.19.90-24.4.v2101.ky10.aarch64/src/driver/linux_resource/autocompat.h) || (rm -f /usr/yzm/onload-master/build/aarch64_linux-4.19.90-24.4.v2101.ky10.aarch64/src/driver/linux_resource/autocompat.h && false)
/usr/yzm/onload-master/src/driver/linux_net/drivers/net/ethernet/sfc/Makefile:95: Kernel does not support vDPA
/usr/yzm/onload-master/src/driver/linux_net/drivers/net/ethernet/sfc/Makefile:97: => vDPA functionality will be disabled
  UPD     /usr/yzm/onload-master/build/aarch64_linux-4.19.90-24.4.v2101.ky10.aarch64/src/driver/linux_net/drivers/net/ethernet/sfc/config.h
echo "  GENERATE /usr/yzm/onload-master/build/aarch64_linux-4.19.90-24.4.v2101.ky10.aarch64/src/lib/ciul/efch_intf_ver.h"
  GENERATE /usr/yzm/onload-master/build/aarch64_linux-4.19.90-24.4.v2101.ky10.aarch64/src/lib/ciul/onload_version.h
T=$(mktemp -p /usr/yzm/onload-master/build/aarch64_linux-4.19.90-24.4.v2101.ky10.aarch64/src/lib/ciul/.); \
(cd /usr/yzm/onload-master/src; /usr/yzm/onload-master/src/../scripts/onload_version_gen) > $T; \
if diff -q /usr/yzm/onload-master/build/aarch64_linux-4.19.90-24.4.v2101.ky10.aarch64/src/lib/ciul/onload_version.h $T 2>/dev/null; \
        then echo "Not updating onload_version.h"; \
        else echo "Updated onload_version.h"; \
             mv $T /usr/yzm/onload-master/build/aarch64_linux-4.19.90-24.4.v2101.ky10.aarch64/src/lib/ciul/onload_version.h; \
fi; \
rm -f $T
  GENERATE /usr/yzm/onload-master/build/aarch64_linux-4.19.90-24.4.v2101.ky10.aarch64/src/lib/ciul/efch_intf_ver.h
  GENERATE /usr/yzm/onload-master/build/aarch64_linux-4.19.90-24.4.v2101.ky10.aarch64/src/lib/transport/ip/uk_intf_ver.h
  CC      /usr/yzm/onload-master/build/aarch64_linux-4.19.90-24.4.v2101.ky10.aarch64/src/lib/citools/buddy.o
  CC      /usr/yzm/onload-master/build/aarch64_linux-4.19.90-24.4.v2101.ky10.aarch64/src/lib/ciul/checksum.o
  CC      /usr/yzm/onload-master/build/aarch64_linux-4.19.90-24.4.v2101.ky10.aarch64/src/lib/citools/buffer.o
  CC      /usr/yzm/onload-master/build/aarch64_linux-4.19.90-24.4.v2101.ky10.aarch64/src/lib/transport/ip/active_wild.o
  CC      /usr/yzm/onload-master/build/aarch64_linux-4.19.90-24.4.v2101.ky10.aarch64/src/lib/ciul/ef100_event.o
  CC      /usr/yzm/onload-master/build/aarch64_linux-4.19.90-24.4.v2101.ky10.aarch64/src/lib/citools/bufrange.o
  CC      /usr/yzm/onload-master/build/aarch64_linux-4.19.90-24.4.v2101.ky10.aarch64/src/lib/transport/ip/common_sockopts.o
  CC [M]  /usr/yzm/onload-master/build/aarch64_linux-4.19.90-24.4.v2101.ky10.aarch64/src/driver/linux_resource/resource_driver.o
  CC      /usr/yzm/onload-master/build/aarch64_linux-4.19.90-24.4.v2101.ky10.aarch64/src/lib/ciul/ef100_vi.o
  CC      /usr/yzm/onload-master/build/aarch64_linux-4.19.90-24.4.v2101.ky10.aarch64/src/lib/citools/copy_iovec.o
  CC      /usr/yzm/onload-master/build/aarch64_linux-4.19.90-24.4.v2101.ky10.aarch64/src/lib/transport/ip/cplane_ops.o
  CC [M]  /usr/yzm/onload-master/build/aarch64_linux-4.19.90-24.4.v2101.ky10.aarch64/src/driver/linux_resource/iopage.o
  CC      /usr/yzm/onload-master/build/aarch64_linux-4.19.90-24.4.v2101.ky10.aarch64/src/lib/ciul/ef10_event.o
  CC      /usr/yzm/onload-master/build/aarch64_linux-4.19.90-24.4.v2101.ky10.aarch64/src/lib/citools/copy_to_iovec.o
  CC      /usr/yzm/onload-master/build/aarch64_linux-4.19.90-24.4.v2101.ky10.aarch64/src/lib/transport/ip/csum_copy_iovec_setlen.o
  CC [M]  /usr/yzm/onload-master/build/aarch64_linux-4.19.90-24.4.v2101.ky10.aarch64/src/driver/linux_resource/kernel_proc.o
  CC      /usr/yzm/onload-master/build/aarch64_linux-4.19.90-24.4.v2101.ky10.aarch64/src/lib/ciul/ef10_evtimer.o
  CC      /usr/yzm/onload-master/build/aarch64_linux-4.19.90-24.4.v2101.ky10.aarch64/src/lib/citools/cpu_features.o
  CC      /usr/yzm/onload-master/build/aarch64_linux-4.19.90-24.4.v2101.ky10.aarch64/src/lib/transport/ip/eplock_slow.o
  CC [M]  /usr/yzm/onload-master/build/aarch64_linux-4.19.90-24.4.v2101.ky10.aarch64/src/driver/linux_resource/filter.o
  CC      /usr/yzm/onload-master/build/aarch64_linux-4.19.90-24.4.v2101.ky10.aarch64/src/lib/citools/crc16.o
[... 495 lines cut ...]
  CC [M]  /usr/yzm/onload-master/build/aarch64_linux-4.19.90-24.4.v2101.ky10.aarch64/src/driver/linux_net/drivers/net/ethernet/sfc/mcdi_mon.o  CC [M]  /usr/yzm/onload-master/build/aarch64_linux-4.19.90-24.4.v2101.ky10.aarch64/src/driver/linux_net/drivers/net/ethernet/sfc/ptp.o
  CC [M]  /usr/yzm/onload-master/build/aarch64_linux-4.19.90-24.4.v2101.ky10.aarch64/src/driver/linux_net/drivers/net/ethernet/sfc/mtd.o
  CC [M]  /usr/yzm/onload-master/build/aarch64_linux-4.19.90-24.4.v2101.ky10.aarch64/src/driver/linux_net/drivers/net/ethernet/sfc/debugfs.o
  CC [M]  /usr/yzm/onload-master/build/aarch64_linux-4.19.90-24.4.v2101.ky10.aarch64/src/driver/linux_net/drivers/net/ethernet/sfc/ioctl.o
  CC [M]  /usr/yzm/onload-master/build/aarch64_linux-4.19.90-24.4.v2101.ky10.aarch64/src/driver/linux_net/drivers/net/ethernet/sfc/ioctl_common.o
  CC [M]  /usr/yzm/onload-master/build/aarch64_linux-4.19.90-24.4.v2101.ky10.aarch64/src/driver/linux_net/drivers/net/ethernet/sfc/kernel_compat.o
  CC [M]  /usr/yzm/onload-master/build/aarch64_linux-4.19.90-24.4.v2101.ky10.aarch64/src/driver/linux_net/drivers/net/ethernet/sfc/sfctool.o
  CC [M]  /usr/yzm/onload-master/build/aarch64_linux-4.19.90-24.4.v2101.ky10.aarch64/src/driver/linux_net/drivers/net/ethernet/sfc/ef100_dump.o
  CC [M]  /usr/yzm/onload-master/build/aarch64_linux-4.19.90-24.4.v2101.ky10.aarch64/src/driver/linux_net/drivers/net/ethernet/sfc/dump.o
  CC [M]  /usr/yzm/onload-master/build/aarch64_linux-4.19.90-24.4.v2101.ky10.aarch64/src/driver/linux_net/drivers/net/ethernet/sfc/driverlink.o
make[3]: *** [scripts/Makefile.build:544：/usr/yzm/onload-master/build/aarch64_linux-4.19.90-24.4.v2101.ky10.aarch64/src/driver/linux_resource] 错误 2
  LD [M]  /usr/yzm/onload-master/build/aarch64_linux-4.19.90-24.4.v2101.ky10.aarch64/src/driver/linux_net/drivers/net/ethernet/sfc/sfc_driverlink.o
  LD [M]  /usr/yzm/onload-master/build/aarch64_linux-4.19.90-24.4.v2101.ky10.aarch64/src/driver/linux_net/drivers/net/ethernet/sfc/sfc.o
[... cut ...]
make: 离开目录“/usr/yzm/onload-master/build/aarch64_linux-4.19.90-24.4.v2101.ky10.aarch64”
onload_build: ERROR: Failed to build driver components.
onload_install: ERROR: Build failed.  Not installing.
[root@localhost scripts]# cat /proc/version
Linux version 4.19.90-24.4.v2101.ky10.aarch64 (KYLINSOFT@localhost.localdomain) (gcc version 7.3.0 (GCC)) #1 SMP Mon May 24 14:45:37 CST 2021
[root@localhost scripts]# uname -a
Linux localhost.localdomain 4.19.90-24.4.v2101.ky10.aarch64 #1 SMP Mon May 24 14:45:37 CST 2021 aarch64 aarch64 aarch64 GNU/Linux

ivatet-amd, 2024-04-25:
Hi @FrankYee2024, thank you for the issue report.

What is the page size in your aarch64 system? (If it is 64K, it's not yet supported by OpenOnload, but we are open for contributions!)

FrankYee2024, 2024-04-25:
Hi ivatet-amd:
    Thank you for your reply.Yes, Our system page size 64K.
[root@localhost scripts]# getconf PAGESIZE
65536

ivatet-amd, 2024-04-25:
Thank you for confirming. Let's keep it open. There is also ON-15679 reported internally.

FranKLeaser, 2024-05-20:
1：I get the kernel source code from https://mirrors.edge.kernel.org/pub/linux/kernel/v4.x/，and the kernel version is 4.19.90, modifying page sizes to 4K and then compile and install the kernel.
[root@localhost ~]# uname -a
Linux localhost.localdomain 4.19.90 #1 SMP Mon May 13 18:52:10 CST 2024 aarch64 aarch64 aarch64 GNU/Linux
[root@localhost ~]# getconf PAGESIZE
4096

2：Install the 8.1.2.26 driver on the new kernel.
[root@localhost ~]# cd /usr/yzm/onload-8.1.2.26/
[root@localhost onload-8.1.2.26]# ls
build  ChangeLog  LICENSE  LICENSES-ALL  mk  README  README.ef_vi  ReleaseNotes  ReleaseNotes-sfc  scripts  src
[root@localhost onload-8.1.2.26]# cd scripts/
[root@localhost scripts]# ls
check_header_presence   check_prototype  mmake           mmaketool  onload_apps   onload_install   onload_misc      onload_rpmtest  onload_tool         sfcaffinity_config  sfcirqaffinity  shell-fns
check_library_presence  libc_compat.sh   mmakebuildtree  onload     onload_build  onload_iptables  onload_profiles  onload_tcpdump  onload_version_gen  sfcaffinity.py      sfcmask.py
[root@localhost scripts]# ./onload_install

3：The error message is as attachment.

[root@localhost scripts]# 
[root@localhost ~]# cat /proc/version
Linux version 4.19.90 (root@localhost.localdomain) (gcc version 7.3.0 (GCC)) #1 SMP Mon May 13 18:52:10 CST 2024
[root@localhost ~]#

FranKLeaser, 2024-05-20:
$cppflags -c /usr/yzm/onload-8.1.2.26/src/tools/cplane/epoll.c -o epoll.o
nice cc -I. -I./../../include -I/usr/yzm/onload-8.1.2.26/src/include -I/usr/yzm/onload-8.1.2.26/src/tools/cplane   -DNDEBUG -D_GNU_SOURCE   -march=native -mtune=native -DAARCH64_PAGE_SIZE=4096  -Wall -Wundef -Wpointer-arith -Wstrict-prototypes -Wnested-externs -Wimplicit-fallthrough=5 -fomit-frame-pointer -DTRANSPORT_CONFIG_OPT_HDR='<ci/internal/transport_config_opt_extra.h>' -fno-strict-aliasing -include /usr/yzm/onload-8.1.2.26/build/gnu_aarch64/cp_intf_ver.h -O2 -g $cflags $cppflags -c /usr/yzm/onload-8.1.2.26/src/tools/cplane/agent.c -o agent.o
nice cc -I. -I./../../include -I/usr/yzm/onload-8.1.2.26/src/include -I/usr/yzm/onload-8.1.2.26/src/tools/cplane   -DNDEBUG -D_GNU_SOURCE   -march=native -mtune=native -DAARCH64_PAGE_SIZE=4096  -Wall -Wundef -Wpointer-arith -Wstrict-prototypes -Wnested-externs -Wimplicit-fallthrough=5 -fomit-frame-pointer -DTRANSPORT_CONFIG_OPT_HDR='<ci/internal/transport_config_opt_extra.h>' -fno-strict-aliasing -include /usr/yzm/onload-8.1.2.26/build/gnu_aarch64/cp_intf_ver.h -O2 -g $cflags $cppflags -c /usr/yzm/onload-8.1.2.26/src/tools/cplane/client.c -o client.o
/usr/yzm/onload-8.1.2.26/src/tools/cplane/server.c:13:10: 致命错误：sys/capability.h：没有那个文件或目录
 #include <sys/capability.h>
          ^~~~~~~~~~~~~~~~~~
编译中断。
make[2]: *** [/usr/yzm/onload-8.1.2.26/mk/after.mk:151：server.o] 错误 1
make[2]: *** 正在等待未完成的任务....
make[2]: 离开目录“/usr/yzm/onload-8.1.2.26/build/gnu_aarch64/tools/cplane”
make[1]: *** [/usr/yzm/onload-8.1.2.26/src/tools/mmake.mk:20：all] 错误 2
make[1]: 离开目录“/usr/yzm/onload-8.1.2.26/build/gnu_aarch64/tools”
make: *** [/usr/yzm/onload-8.1.2.26/src/mmake.mk:25：all] 错误 2
make: 离开目录“/usr/yzm/onload-8.1.2.26/build/gnu_aarch64”
onload_build: ERROR: Failed to build 64-bit user-level components.
onload_install: ERROR: Build failed.  Not installing.
[root@localhost scripts]#

ivatet-amd, 2024-05-20:
Hi @FranKLeaser, for the missing "sys/capability.h", please try installing the `libcap-devel` or `libcap-dev` package depending on your distro.