Xilinx-CNS/onload #223: epoll spinning breaks epoll+accept+reuesport

romandivackyabsa, 2024-05-24:
I am using Onload 8.1.2.26 and I am hitting an issue where asynchronous accept (via epoll) is not returning with reuseport socket + epoll spinning.  I've attached a testcase that reproduces this (prints ACCEPTED when something connects to the port 12345), I am using it with this profile:

onload_set EF_POLL_USEC 6000000          # commenting this out makes it work
onload_set EF_CLUSTER_IGNORE 1

The testcase works fine when I remove any of these 3 things: 1) reuseport 2) epoll spinning 3) calling the asio::async_accept from different thread (the commented out ios.dispatch() in the testcase).  

Am I doing anything wrong? Is this a bug in onload? I dont use clustering so it seems suspicious to me that commenting out reuseport makes things work, also dont see why spinning should make any difference.

```
#include <boost/asio.hpp>
#include <boost/asio/io_service.hpp>

#include <iostream>
#include <thread>
#include <sys/socket.h>

int main(int argc, char** argv)
{
    using namespace boost::asio;
    io_service ios;
    ip::tcp::acceptor acceptor(ios);
    ip::tcp::socket   socket(ios);

    acceptor.open(ip::tcp::v4());

    int v = 1;
    setsockopt(acceptor.native_handle(), SOL_SOCKET, SO_REUSEPORT, (char*)&v, sizeof(v));    // comment out to make it work

    boost::system::error_code ec;
    acceptor.bind(*ip::tcp::resolver(ios).resolve(ip::tcp::resolver::query("192.168.70.21", "12345")), ec);
    acceptor.listen(socket_base::max_connections, ec);

    auto t = std::thread([&](){  ios.run(); });

    //ios.dispatch([&]() {    // uncomment to make it work
            acceptor.async_accept(socket, [](const boost::system::error_code& ec) { std::cout << "ACCEPTED\n"; });
     //       });

    sleep(3600);
}
```

romandivackyabsa, 2024-05-28:
Also, setting EF_UL_EPOLL=3 fixes the issue too.

ivatet-amd, 2024-05-29:
Hi @romandivackyabsa,

Is there a race condition between invoking `ios.run()` in a separate thread and `acceptor.async_accept()`? It seems like `ios.run()` provides executable context for `acceptor`, and if `ios.run()` manages to make enough progress before `acceptor.async_accept()`, it concludes there is nothing to do and exits early, leaving no executable context to invoke the accept hander/lambda.

The code snippet you attached indeed works for me as is (built with `g++ -g -lpthread -lboost_system test.cpp` in RHEL8.6), but as soon as I try to run it either with Onload and `EF_POLL_USEC=6000000`, or with `strace` or `gdb`, the issue reproduces, i.e. the app doesn't print the happy `ACCEPTED` message. This behaviour makes me suspect a timing-sensitive race condition.

A fix that works for me is to start a new thread with `ios.run()` after `acceptor.async_accept()`:

```c++
    //ios.dispatch([&]() {    // uncomment to make it work
            acceptor.async_accept(socket, [](const boost::system::error_code& ec) { std::cout << "ACCEPTED\n"; });
     //       });

    auto t = std::thread([&](){  ios.run(); });
```

Please let me know what you think. I'm unfamiliar with the Boost library, so please take the above explanation with a grain of salt.

romandivackyabsa, 2024-05-29:
Hi Ivan,

Thanks for your response. I think you are right, I tried to make a small example reproducing the problem from our (big) application but made this silly mistake. The problem in the big application is not this one though.

The big application in fact listens on two sockets/ports, clients try to connect to both in a short succession (within a second). The first one succeeds, but the second accept handler never gets called (I think because the first accept puts onload into spinning mode). What I wrote in my first report holds. Removing any of reuseport/different-thread/spinning makes it work (ie. both connections get accepted).

The epoll thread has this backtrace:
```
Thread 1 (Thread 0x7fa8291b8700 (LWP 664440)):
#0  0x00007fa83a7247cb in ioctl () from /lib64/libc.so.6
#1  0x00007fa83cd441d7 in citp_epoll_wait (fdi=fdi@entry=0x607000011b30, events=events@entry=0x7fa8291b6030, ordering=ordering@entry=0x0, maxevents=maxevents@entry=128, timeout_hr=<optimized out>, sigmask=sigmask@entry=0x0, lib_context=0x7fa8291b5e70) at /var/lib/dkms/onload/8.1.2.26/build/src/lib/transport/unix/epoll_fd.c:2146
#2  0x00007fa83cd262fa in onload_epoll_wait (epfd=20, events=0x7fa8291b6030, maxevents=128, timeout=-1) at /var/lib/dkms/onload/8.1.2.26/build/src/lib/transport/unix/ul_epoll.h:276
#3  0x000000000167a602 in boost::asio::detail::epoll_reactor::run (this=0x611000027140, usec=-1, ops=...) at /usr/include/boost/asio/detail/impl/epoll_reactor.ipp:471
#4  0x000000000167d24c in boost::asio::detail::scheduler::do_run_one (this=0x6110000138c0, lock=..., this_thread=..., ec=...) at /usr/include/boost/asio/detail/impl/scheduler.ipp:385
#5  0x000000000167cac5 in boost::asio::detail::scheduler::run (this=0x6110000138c0, ec=...) at /usr/include/boost/asio/detail/impl/scheduler.ipp:154
#6  0x000000000167df65 in boost::asio::io_context::run (this=0x60600004ef48) at /usr/include/boost/asio/impl/io_context.ipp:62
```
Onload_stackdump lots shows me this:

```
(the correctly accepted connection)
TCP 19:2043 lcl=192.168.70.21:16575 rmt=192.168.70.11:51822 ESTABLISHED
  lock: 0
  rx_wake=00000067(RQ) tx_wake=00000040(RQ) flags: O_NONBLOCK O_NDELAY
  ul_poll: 2095004000 spin cycles 1000000 usec
  uid=1006 s_flags: TCP_NODELAY CONNECT_MUST_BIND PMTU_DO
  rcvbuf=131400 sndbuf=131072
  rcvtimeo_ms=0 sndtimeo_ms=0 sigown=0 cmsg=
  bindtodev=0(0,0x0:0) ttl=-1 BOUND_ADDR
  rx_errno=0 tx_errno=0 so_error=0 os_sock=0,TX
  TX timestamping queue: q_pkts=0 reap=0 tot_pkts=0
  tcpflags: WSCL ESTAB PASSIVE  local_peer: -1
  snd: up=6c1218ad una-nxt-max=6c1218ad-6c1218ad-6c1314ad enq=6c1218ad
  snd: send=0(0) send+pre=0 inflight=0(0) wnd=64512 unused=64512
  snd: cwnd=3366+0 used=0 ssthresh=33553920 bytes_acked=0 Open
  snd: timed_seq 6c1218a6 timed_ts 94dab48e
  snd: sndbuf_pkts=135 Onloaded(Valid) if=9 mtu=1500 intf_i=4 vlan=0 encap=0 verinfo 21e-7
  snd: limited rwnd=0 cwnd=0 nagle=0 more=0 app=63
  rcv: nxt-max=501e8c15-50208d56 wnd adv=131393 cur=131400  FASTSTART FAST
  rcv: isn=501e891c up=501e891b urg_data=0000 q=recv1
  rcv: bytes=761 tot_pkts=a6 rob_pkts=0 q_pkts=2+0 usr=0
  eff_mss=1460 smss=1460 amss=1460  used_bufs=2 wscl s=9 r=2
  srtt=01 rttvar=000 rto=200 zwins=0,0
  curr_retrans=0 total_retrans=0 dupacks=0 congrecover=0
  rtos=0 frecs=0 seqerr=0,0 ooo_pkts=0 ooo=0
  tx: defer=0 nomac=0 warm=0 warm_aborted=0
  tmpl: send_fast=0 send_slow=0 active=0
  timers:
------------------------------------------------------------
TCP 19:2045 lcl=192.168.70.21:16575 rmt=0.0.0.0:0 LISTEN
  lock: 0
  rx_wake=00000001(RQ) tx_wake=00000000(RQ) flags: O_NONBLOCK O_NDELAY OS_BACKED
  ul_poll: 2095004000 spin cycles 1000000 usec
  uid=1006 s_flags: FILTER BOUND PBOUND REUSEPORT PMTU_DO
  rcvbuf=87380 sndbuf=16384
  rcvtimeo_ms=0 sndtimeo_ms=0 sigown=0 cmsg=
  bindtodev=0(0,0x0:0) ttl=-1 BOUND_ADDR
  rx_errno=6b tx_errno=20 so_error=0 os_sock=0,TX
  listenq: max=2048 n=0 new=0 buckets=1
  acceptq: max=128 n=0 accepted=1
  defer_accept=255
  sockcache: n=0 sock_n=0 cache=EMPTY pending=EMPTY connected=EMPTY
  sockcache_hit=0
  l_overflow=0 l_no_synrecv=0 aq_overflow=0 aq_no_sock=0 aq_no_pkts=0
  a_loop2_closed=0 a_no_fd=0 ack_rsts=0 os=0 rx_pkts=2

(the not accepted connection)
TCP 19:2042 lcl=192.168.70.21:16576 rmt=192.168.70.11:37996 CLOSE-WAIT
  lock: 0
  rx_wake=00000002     tx_wake=00000001     flags: ORPH ACCEPTQ
[... 19 lines cut ...]
  curr_retrans=0 total_retrans=0 dupacks=0 congrecover=0
  rtos=0 frecs=0 seqerr=0,0 ooo_pkts=0 ooo=0
  tx: defer=0 nomac=0 warm=0 warm_aborted=0
  tmpl: send_fast=0 send_slow=0 active=0
  timers:
------------------------------------------------------------
TCP 19:2044 lcl=192.168.70.21:16576 rmt=0.0.0.0:0 LISTEN
  lock: 0
  rx_wake=00000001     tx_wake=00000000     flags: O_NONBLOCK O_NDELAY OS_BACKED
  ul_poll: 2095004000 spin cycles 1000000 usec
  uid=1006 s_flags: FILTER BOUND PBOUND REUSEPORT PMTU_DO
  rcvbuf=87380 sndbuf=16384
  rcvtimeo_ms=0 sndtimeo_ms=0 sigown=0 cmsg=
  bindtodev=0(0,0x0:0) ttl=-1 BOUND_ADDR
  rx_errno=6b tx_errno=20 so_error=0 os_sock=0,TX
  listenq: max=2048 n=0 new=0 buckets=1
  acceptq: max=128 n=1 accepted=0
  defer_accept=255
  sockcache: n=0 sock_n=0 cache=EMPTY pending=EMPTY connected=EMPTY
  sockcache_hit=0
  l_overflow=0 l_no_synrecv=0 aq_overflow=0 aq_no_sock=0 aq_no_pkts=0
  a_loop2_closed=0 a_no_fd=0 ack_rsts=0 os=0 rx_pkts=2
```

Unfortunately I am unable to distill a small example that reproduces this. Any idea what can be wrong?

ivatet-amd, 2024-05-29:
@romandivackyabsa,

Could you try running the Onloaded app with `EF_EPOLL_CTL_FAST=0`? This optimisation is enabled by default and defers the `epoll_ctl()` invocations until `epoll_wait()`. In rare cases, it might cause inconsistency in what an Onloaded app thinks and what is submitted to Onload for polling.

We think this might be the case because there is a connection waiting to be accepted (`n=1`):
```
TCP 19:2044 lcl=192.168.70.21:16576 rmt=0.0.0.0:0 LISTEN
  ...
  acceptq: max=128 n=1 accepted=0
```

... while the app is blocked in `onload_epoll_wait()` as per backtrace (assuming it does the right thing).

ivatet-amd, 2024-05-29:
Another contender is `EF_FDS_MT_SAFE=0`, which can be tried alone or with the previous option.

ivatet-amd, 2024-05-29:
Overall, there is the "safe" Onload profile at [scripts/onload_profiles/safe.opf](https://github.com/Xilinx-CNS/onload/blob/master/scripts/onload_profiles/safe.opf), which can also mitigate issues by disabling optimisations based on the assumptions incorrect for multithreaded Boost C++ apps.

romandivackyabsa, 2024-05-29:
EF_EPOLL CTL_FAST=0 makes it work. But so does EF_UL_EPOLL=3, why would that be the case? Also, in the docs describing EF_EPOLL_CTL_FAST I read:

>This can break applications that do
>epoll_wait() in one thread and epoll_ctl() in another thread. The issue
>only affects EF_UL_EPOLL=2 and the solution is to set EF_EPOLL_CTL_FAST=0
>if this is a problem. The described condition does not occur if EF_UL_EPOLL=1
>or EF_UL_EPOLL=3.

I also managed to distill a small test case that reproduces the issue.

```
#include <boost/asio.hpp>
#include <boost/asio/io_service.hpp>

#include <iostream>
#include <thread>
#include <sys/socket.h>

thread_local static std::thread s_thread;

int main(int argc, char** argv)
{
    using namespace boost::asio;
    io_service ios;

    std::unique_ptr<boost::asio::io_service::work> work = std::make_unique<boost::asio::io_service::work>(ios);

    auto t = std::thread([&](){
            s_thread = std::thread([]() { sleep(3600); });
            ios.run();
            });

    ip::tcp::acceptor acceptor1(ios), acceptor2(ios);
    acceptor1.open(ip::tcp::v4());

    int v = 1;
    setsockopt(acceptor1.native_handle(), SOL_SOCKET, SO_REUSEPORT, (char*)&v, sizeof(v));

    boost::system::error_code ec;
    acceptor1.bind(*ip::tcp::resolver(ios).resolve(ip::tcp::resolver::query("192.168.70.21", "12345")), ec);
    acceptor1.listen(socket_base::max_connections, ec);

    ip::tcp::socket   socket1(ios), socket2(ios);
    acceptor2.open(ip::tcp::v4());

    setsockopt(acceptor2.native_handle(), SOL_SOCKET, SO_REUSEPORT, (char*)&v, sizeof(v));

    acceptor2.bind(*ip::tcp::resolver(ios).resolve(ip::tcp::resolver::query("192.168.70.21", "12346")), ec);
    acceptor2.listen(socket_base::max_connections, ec);

    //ios.dispatch([&]() {
            acceptor1.async_accept(socket1, [&](const boost::system::error_code& ec) {
                    socket1.set_option(boost::asio::ip::tcp::no_delay(true));
                    socket1.non_blocking(true);

                    std::cout << "ACCEPTED 1\n";
                    });
            acceptor2.async_accept(socket2, [&](const boost::system::error_code& ec) {
                    socket2.set_option(boost::asio::ip::tcp::no_delay(true));
                    socket2.non_blocking(true);

                    std::cout << "ACCEPTED 2\n";
                    });
     //       });

    sleep(3600);
}
```

Initializing the s_thread in the ios thread seems to make a difference. I suspect it changes some timing.

romandivackyabsa, 2024-05-29:
Actually, I spoke too soon. EF_EPOLL CTL_FAST=0 fixes the small testcase I posted, but not the big application. EF_UL_EPOLL=3 does. What is the actual difference between EF_UL_EPOLL=1 and 3?

ivatet-amd, 2024-05-29:
Modes 1 and 3 for `EF_UL_EPOLL` make a difference for larger fd sets, and I cannot see why they impact your app(s) if they operate on smaller fd sets.

I agree with you that it might change timings in your app, and therefore suspect a nasty race condition. Try narrowing it down further?

romandivackyabsa, 2024-05-30:
I noticed something interesting, I some timestamping and reading of data after establishing the connection (the testcase attached).

With EF_EPOLL_CTL_FAST=0 the second connection gets accepted but with a delay. The delay seems to scale with the value of EF_POLL_USEC until it's over 1 second. Ie. with 600ms of spinning the second connection gets accepted 600ms after the first one.

ie.

```
cat fom-latency.opf
onload_set EF_POLL_USEC 600000
onload_set EF_CLUSTER_IGNORE 1
#onload_set EF_UL_EPOLL 3
onload_set EF_EPOLL_CTL_FAST 0
#onload_set EF_EPOLL_MT_SAFE 0

onload -p ./fom-latency.opf ./a.out
onload: EF_CLUSTER_IGNORE is deprecated use EF_CLUSTER_SIZE instead
oo:a.out[1189313]: Using Onload 8.1.2.26 [34]
oo:a.out[1189313]: Copyright (c) 2002-2023 Advanced Micro Devices, Inc.
ACCEPTED 1 2024-05-30T10:00:05.483352Z
READ 1 7
ACCEPTED 2 2024-05-30T10:00:06.083903Z
READ 2 7
```

when on the client box I do:

```echo foobar | nc 192.168.70.21 12345& echo foobar | nc 192.168.70.21 12346 &.```

The delay stops maxes out at 1 second. This should be easily reproducible for you. Can you explain that?

ivatet-amd, 2024-05-30:
I re-ran the app from the latest snippet with additional Onload logging `EF_LOG_TIMESTAMPS=1 EF_LOG_THREAD=1 EF_UNIX_LOG=0x4000` and reproduced your observation too. However, it made a series of syscalls, including `epoll_ctl()`, from different threads around the same time, potentially causing Onload descriptor table inconsistencies. If so, `EF_FDS_MT_SAFE=0` (set by default) should work for you. Other configuration options seem irrelevant at the moment and make us rely on their implementation details.

romandivackyabsa, 2024-05-30:
Indeed, adding EF_FDS_MT_SAFE fixes my issue, I originally misread EF_FDS_ as EF_EPOLL_. I still find the delay rather surprising. Do I need EF_EPOLL_MT_SAFE=0 too?  I do call epoll_ctl and epoll_wait from different threads but not setting that one seems to be fine

Just for the record, I also found that the delay is the cause of the big application issues. The EF_EPOLL_CTL_FAST=0 does help even the big app, but only when the spinning time is low (ie. 100ms or so). What happens is that with the 6 second spinning (and after uncommenting the recursive readSome call so it keeps reading) the delay grows even over 1 second and the client (during 3way handshake) simply decides to send a FIN after 1 second as it doesnt receive any answers. I guess onload then decides that it's pointless to return from the accept as the connection has terminated.

ivatet-amd, 2024-05-30:
The `EF_EPOLL_MT_SAFE` is unset by default, so keep it unchanged. Apparently, `EF_FDS_MT_SAFE=0` is everything you need for the correct operation. (Indeed, these two have pretty similar names!)

A theory based on my observations:

- The `epoll_ctl()` invocations keep the Onload kernel in sync with the actual epoll set, i.e. what the application thinks. (Thanks to the non-default `EF_EPOLL_CTL_FAST=0`, which removes optimisations.)
- The `epoll_wait()` invocations spin in the Onload userspace which might be out-of-sync due to unsafe concurrency handling. Consequnetly, the app is not seeing new clients. (The unsafety is caused by the default `EF_FDS_MT_SAFE=1`, which we need to unset.)
- Eventually (after `EF_POLL_USEC=6000000`), the Onload userspace invokes the Onload kernelspace, which returns "successfully" immediately because it is in sync with the epoll set. Finally, the server sees a new client.

AFAIU, if the three-way handshake has occurred, the kernel and Onload must tell the app about it even if `accept()` happens after receiving FIN.

ivatet-amd, 2024-05-31:
I should have clarified the previous comment: A theory based on my observations, although it might be different given you are running bigger apps with possibly different behaviour.

@romandivackyabsa Have you seen further issues with `EF_FDS_MT_SAFE=0`?

ivatet-amd, 2024-06-04:
The issues you observed are root-caused by the limitations of `onload_move_fd()`. (The Onload User Guide has more information about it at https://docs.amd.com/r/en-US/ug1586-onload-user/onload_move_fd.) Specifically, `onload_move_fd()` should not operate on file descriptors moved across stacks. Otherwise, Onload inadvertently removes them from their epoll set. We track it internally as ON-15835.

Now, your application operates on the same file descriptors (listening sockets) from different threads. In some cases, like deferring the `epoll_ctl()` invocations from one thread to another, Onload shares the workflow with the `onload_move_fd()` handling and removes file descriptors from their epoll sets. As a result, your application doesn't get control back from `epoll_wait()` even though there is a pending client ready to be `accept()`'ed.

The epoll mode 3 in Onload has a different (from mode 1) runtime profile/behaviour. Consequently, this mode mitigates the above situation altogether or re-adds file descriptors to their I/O multiplexes, also called "reprobe" in the code base, and recovers the affected epoll set. The same effects apply to the other workarounds we discussed above, `EF_FDS_MT_SAFE=0` and `EF_EPOLL CTL_FAST=0`. They can change the Onload workflow in favour of your app and address the issue.

romandivackyabsa, 2024-06-06:
Ok. Thanks for the explanation!

ivatet-amd, 2024-06-06:
Thank you @romandivackyabsa. Please email our support team at support-nic@amd.com if you need further help with the Onload and/or hardware tuning for your application. 👍

closed 2024-06-06 by ivatet-amd