Xilinx-CNS/onload #224: [sfc efrm] init_sfc_resource: ERROR: failed to find syscall table

tsatsulin, 2024-05-27:
Hi guys!

On Ubuntu 22.04 with the kernel 5.15.0-107-generic I faced the following issue:

```
modprobe: ERROR: could not insert 'onload': Unknown symbol in module, or unknown parameter (see dmesg)
onload_tool: FAILED: /sbin/modprobe onload
```

`dmesg` shows the following:

```
[ 1382.474726] [sfc efrm] init_sfc_resource: ERROR: failed to find syscall table
```

Can you please suggest if this is something already known or a new issue?

tsatsulin, 2024-05-27:
I can also see the following:

```
openonload/8.1.2.26, 5.15.0-107-generic, x86_64: installed (WARNING! Diff between built and installed module!) (WARNING! Diff between built and installed module!)
```

But I removed `onload` using `onload_uninstall` and checked that nothing left after this operation.

On another host where `onload` works, I can see the same:

```
openonload, 8.1.1.17, 5.4.0-169-generic, x86_64: installed (WARNING! Diff between built and installed module!) (WARNING! Diff between built and installed module!)
```

abower-amd, 2024-05-27:
> I can also see the following:
>
> ```
> openonload/8.1.2.26, 5.15.0-107-generic, x86_64: installed (WARNING! Diff between built and installed module!) (WARNING! Diff between built and installed module!) [...]

Am I right in thinking that this is reported by DKMS? There is some overlap between what DKMS and the onload install scripts do - have you checked the correct steps at https://docs.amd.com/r/en-US/ug1586-onload-user/Building-and-Installing-from-a-DKMS-Package ?

If you wish to use DKMS, the current _supported_ approach for Debian is to convert the RPM DKMS package but note that for Debian-based systems there are unreleased DKMS enhancements in the `master` branch. To benefit from these use the `scripts/debian/onload_mksrcdeb.sh` script to generate a debian source package. When you build the debian source package you will get both a kernel module package for the current OS and a DKMS package - you would choose one approach or the other but not both. As this is an unreleased enhancement it is not yet documented.

Alternatively, avoid DKMS and use the onload_install/onload_tool scripts exclusively.

If you have issues with DKMS please raise a *separate* issue as I am about to close this syscall ticket as a duplicate!

Good luck and let us know how you get on if you try the new functionality!

abower-amd, 2024-05-27:
Hi @tsatsulin,

This is a duplicate of #216, which is now resolved.

This error occurs because of new speculative execution mitigations in recent Linux kernels. These mitigations have been back-ported to kernels existing OS releases, such as Debian 12 and Ubuntu 22.04.

Please report issues with supported (numbered) Onload releases to <support-nic@amd.com> and report issues on this github repository that are experienced on the head of the `master` branch that includes recent development work.

Thank you!

closed 2024-05-27 by abower-amd

tsatsulin, 2024-05-27:
Hi @abower-amd,

Thank you for quick turnaround.

> Am I right in thinking that this is reported by DKMS?

Yes, you are right, this is from `dkms status`.

> There is some overlap between what DKMS and the onload install scripts do - have you checked the correct steps at https://docs.amd.com/r/en-US/ug1586-onload-user/Building-and-Installing-from-a-DKMS-Package ?

Thank you for the link. I will double-check if we missed anything from this guide.

I will follow your instructions on reporting new issues in the future. Thanks!

abower-amd, 2024-05-31:
@tsatsulin, The OpenOnload-8.1.3.40 is now available with this fix as a supported solution. If you still have trouble with DKMS installation with an AMD Solarflare NIC and this version then the support team can help you!

tsatsulin, 2024-05-31:
@abower-amd Great news! I will let you know if I see any issues in this release. Thank you for your help!

tsatsulin, 2024-06-25:
This is just a confirmation that everything works fine with the latest version, 8.1.3.40. Thank you for your help!