Xilinx-CNS/onload #229: citp_fdtable_ctor: failed to allocate fdtable

ol-alexandra, 2024-06-13:
Debian testing/sid:
```
oo:ncat[1582]: citp_fdtable_ctor: failed to allocate fdtable (0x3ffffff8)
oo:ncat[1582]: FAIL at /home/sasha/work/level5/onload/src/lib/transport/unix/startup.c:685
oo:ncat[1582]: EtherFabric transport library: failed to initialise (9)
oo:ncat[1582]: hostname=elrond pid=1582
/home/sasha/work/level5/onload/build/gnu_x86_64_debt/lib/transport/unix/libcitransport0.so(+0x109f4d)[0x7f235a08df4d]
/home/sasha/work/level5/onload/build/gnu_x86_64_debt/lib/transport/unix/libcitransport0.so(+0xb4b1)[0x7f2359f8f4b1]
/lib64/ld-linux-x86-64.so.2(+0x4e35)[0x7f235a0f2e35]
/lib64/ld-linux-x86-64.so.2(+0x4f64)[0x7f235a0f2f64]
/lib64/ld-linux-x86-64.so.2(+0x1a840)[0x7f235a108840]
Aborted (core dumped)
```
The issue is, Onload uses `RLIMIT_NOFILE:rlim_max` value as fdtable size.  With the recent changes in systemd the soft limit is the same as previously (1024, select-compatible) but hard limit is increased to 0x3ffffff8.  System can't allocate that amount of memory for fdtable.

okt-sergeyn, 2024-11-11:
I see this issue on Ubuntu 24.10. 
According to my tests, if swapping is on, the issue does not happen.

abower-amd, 2025-11-07:
> The issue is, Onload uses `RLIMIT_NOFILE:rlim_max` value as fdtable size. With the recent changes in systemd the soft limit is the same as previously (1024, select-compatible) but hard limit is increased to 0x3ffffff8.

I wonder if this change got reverted at some point during the trixie development cycle?

I just created a fresh trixie VM, ran `prlimit` from a bash login shell and from a `systemd-run` invocation and got a hard limit of 524288 in both cases. Am I not looking in the right place?

The same with Resolute Raccoon.

closed 2025-11-11 by abower-amd