Xilinx-CNS/onload #231: onload AF_XDP bricks aws ec2

lparkersc, 2024-06-25:
### Environment
- **OS**: RHEL 9
- **Kernel**: 5.14.0-70
- **Instance Type**: m5zn.metal
- **Network Driver**: ENA
- **Commit**: [f8458aa](https://github.com/Xilinx-CNS/onload/commit/f8458aaa4052c130436ffe6d5a110fb72f2e3c92)
- **Branch**: master (as of this writing)

### Issue Description
I am attempting to install OpenOnload configured for XDP on the ENA driver. The setup process runs smoothly until I test the application, at which point my SSH connection is disrupted, and the instance becomes unresponsive.

### Steps to Reproduce
1. Install all required packages.
2. Execute `make`.
3. Run `./onload_install --build-profile cloud` as per the [documentation](https://docs.amd.com/r/en-US/ug1586-onload-user/Cloud-Build-Profile?tocId=l~~5oIDLEQYxqMkHy_G2Sw).
4. Successfully load sfc modules with `sudo onload_tool reload`.
5. Attach a secondary ENA interface (name: eth1, driver: ena) to my EC2 instance.
6. Register the new ENA interface with sfc_resource: `echo eth1 > /sys/module/sfc_resource/afxdp/register`.
7. Disable SELinux and run my test application: `onload ./client.py [server_ip] [port]`.
8. Post-step 7, my SSH connection is disrupted, and I cannot reconnect to the instance.

### Additional Information
I cannot post the system logs as the instance becomes unresponsive. I referenced this [previous discussion](https://github.com/Xilinx-CNS/onload/issues/28) where experiments with OpenOnload on ENA via XDP showed mixed results. The setup process takes about 30 minutes, complicating the troubleshooting process.

I am relatively new to OpenOnload, so any advice on potential missteps or additional configurations would be greatly appreciated. 

Thank you for any assistance you can provide.

maxholloway, 2025-01-12:
You ever find a resolution to this @lparkersc ? I just hit the same bug ☠️

beanqi, 2025-03-11:
I hit the same issue.

- kernel version : 6.8.0-1024-aws

nastynaz, 2026-02-27:
Any updates on this? Did you manage to get it working? @lparkersc @beanqi