Xilinx-CNS/onload #233: zero-copy api UDP transmit

abhinavchadaga, 2024-07-10:
Apologies if this has already been answered or is otherwise obvious but does the zero-copy api support UDP transmit packet data? The docs only mention UDP received packet data and TCP transmit packet data and [this](https://stackoverflow.com/questions/15796937/why-openonload-extensions-api-for-zero-copy-doesnt-have-anything-for-udp-send-o) StackOverflow post seems to indicate that support was not present 11 years ago. The documentation for `onload_zc_send` in `extensions_zc.h` seems to indicate that UDP can be used but there is a TODO at the top of the same file for zero-copy UDP-TX.

I'd like to use UDP zero-copy transmits for something I'm working on so if there's support for it, that would be good to know.

abhinavchadaga, 2024-07-10:
Thanks for the response. Which PR are you referring to though? I don't see a PR# higher than 232?

rhughes-xilinx, 2024-07-10:
Oops, sorry, that's a reference to a different repo. The API docs are in src/include/cplane/api.h

abhinavchadaga, 2024-07-10:
awesome thank you! closing out the issue!

closed 2024-07-10 by abhinavchadaga

rhughes-xilinx, 2024-07-11:
FYI the sample app I mentioned from that other repo is now in this repo: `src/tests/ef_vi/efsend_cplane.c`