Xilinx-CNS/onload #234: Multicast loopback not working when payload is higher than MTU

atik7, 2024-07-23:
how to receive loopback multicast when payload is bigger than MTU?

EF_MCAST_SEND=3.
EF_MCAST_RECV_HW_LOOP=1.

btw, loopback works only payload is lower than MTU.

ivatet-amd, 2024-07-24:
The assumption is that it is NIC X2 Series with the full-feature Firmware variant, multiple Onloaded applications that send and receive the multicast traffic via accelerated sockets, and MTU cannot be changed.

Currently, we have an unresolved ticket, ON-14652, opened for the above configuration and `EF_MCAST_RECV_HW_LOOP=1 EF_MCAST_SEND=3` so that it is expected that your Onloaded receiver does not receive loopback fragmented multicast.

Meanwhile, could you try the Onload stack sharing with `EF_NAME=foo EF_MCAST_SEND=3` as a workaround? It might work for you.

Further, please email our Support Team at [support-nic@amd.com](mailto:support-nic@amd.com) for help with tuning specifically for your application.

ivatet-amd, 2024-07-24:
However, with the Onload stack sharing `EF_NAME=foo`, please be aware that the affected Onloaded applications might interfere with each other in undesirable ways. For instance, they might end up contending for the same Onload stack lock, reducing their overall performance. The Onload User Guide also warns about it at https://docs.amd.com/r/en-US/ug1586-onload-user/Stack-Sharing.

The suggestion to contact our Support Team is still the case.

atik7, 2024-07-25:
Thank you for the details. Correct, it does work if EF_NAME are same, but this is not we are looking for, we need to support different EF_NAME, is there any priority for ON-14652.

Noted about the support email address,