Xilinx-CNS/onload #286: dkms overrides issue

lukester1975, 2025-06-06:
Last one. Well, two.

1. Installing a dkms deb built from something like`devscripts_build_profiles="pkg.onload.havesdci" ./scripts/onload_mkpackage -D --version ${VER} --resultdir build/${VER} --user --dkms` fails, because the sdci flag gets appended to the MAKE var in `dkms_override.conf` without a separator, leading to:

```
....
Building module:
Cleaning build area......
unset CC; /var/lib/dkms/onload/9.0.2/build/scripts/onload_install --newkernel 6.14.0-15-generic--have-sdci
```
https://github.com/Xilinx-CNS/onload/compare/master...lukester1975:fix-dkms-overrides

2. Similarly to #285 point 2, the generated initrd isn't including the correct sfc module. I spent a little time looking but can't figure it out. `/lib/modules/$(uname -r)/modules.dep` looks correct, no obvious path issue like with akmod:

```
kernel/drivers/net/ethernet/sfc/falcon/sfc-falcon.ko.zst: kernel/drivers/net/mdio.ko.zst kernel/drivers/i2c/algos/i2c-algo-bit.ko.zst kernel/drivers/mtd/mtd.ko.zst
kernel/drivers/net/ethernet/sfc/siena/sfc-siena.ko.zst: kernel/drivers/net/mdio.ko.zst kernel/drivers/mtd/mtd.ko.zst
updates/dkms/sfc_char.ko.zst: updates/dkms/sfc_resource.ko.zst
updates/dkms/onload.ko.zst: updates/dkms/sfc_char.ko.zst updates/dkms/sfc_resource.ko.zst
extra/sfc.ko: kernel/drivers/mtd/mtd.ko.zst
updates/dkms/sfc_resource.ko.zst:
```
But on boot it's the still the stock sfc driver. Is that expected? Seems odd.

It appears `update-initramfs` is ignoring depmod, even though strace showed it was reading it. Or maybe my expectation it would generate the initrd in a similar way to dracut is misplaced (it doesn't apply overrides maybe, it's just using it for dependencies)?

Either way, it might be nice to normalize the behaviour between akmod/dkms if it's possible at this layer. That said, i guess they do behave the same right now for different reasons 😀

FWIW, hacking `MODULES=dep` for the initramfs config avoids the issue, but obviously not the correct solution...

Thanks

jfeather-amd, 2025-06-09:
Thanks, your code change for the debian DKMS package looks good. I agree that your second point seems like an issue - I'd expect onload might have trouble if it's running with the stock sfc driver! Out of curiosity, how are you determining which sfc driver is in use?

lukester1975, 2025-06-09:
The device names are different 😀 (plus `ethtool -i` showing 6.8.0-60-generic vs 6.0.2.1008).

I forgot to paste in the relevant `update-initramfs` bits:

```
update-initramfs -c -k $(uname -r) -v 2>&1 | grep -E "(sfc|dkms|extra|onload)"
dracut-install: mkdir '/var/tmp/mkinitramfs_PPBWdH/lib/modules/6.8.0-60-generic/kernel/drivers/net/ethernet/sfc'
dracut-install: mkdir '/var/tmp/mkinitramfs_PPBWdH/lib/modules/6.8.0-60-generic/kernel/drivers/net/ethernet/sfc/falcon'
dracut-install: cp '/lib/modules/6.8.0-60-generic/kernel/drivers/net/ethernet/sfc/falcon/sfc-falcon.ko.zst' '/var/tmp/mkinitramfs_PPBWdH/lib/modules/6.8.0-60-generic/kernel/drivers/net/ethernet/sfc/falcon/sfc-falcon.ko.zst'
dracut-install: mkdir '/var/tmp/mkinitramfs_PPBWdH/lib/modules/6.8.0-60-generic/kernel/drivers/net/ethernet/sfc/siena'
dracut-install: cp '/lib/modules/6.8.0-60-generic/kernel/drivers/net/ethernet/sfc/siena/sfc-siena.ko.zst' '/var/tmp/mkinitramfs_PPBWdH/lib/modules/6.8.0-60-generic/kernel/drivers/net/ethernet/sfc/siena/sfc-siena.ko.zst'
dracut-install: cp '/lib/modules/6.8.0-60-generic/kernel/drivers/net/ethernet/sfc/sfc.ko.zst' '/var/tmp/mkinitramfs_PPBWdH/lib/modules/6.8.0-60-generic/kernel/drivers/net/ethernet/sfc/sfc.ko.zst'
Adding config /etc/initramfs-tools/conf.d/onload.conf
Adding config /etc/modprobe.d/dkms.conf
Adding config /etc/modprobe.d/onload.conf
```
despite `modules.conf` looking like mentioned above.

(BTW I did wonder if it was some oddity of generating an initrd for the running kernel and the stock driver being loaded, but ensuring the onload sfc module is running and then generating the initrd still exhibits the same behaviour.)

ivatet-amd, 2025-06-24:
Thanks @lukester1975 -- Both issues should be fixed now. 👍

Sadly, `update-initramfs` is not entirely clear in the documentation/manpage about what gets included, so @jfeather-amd and I figured out that the drop-in config file works reliably in our testing, and I hope it is not too implementation-specific.

Another problem was that `dh_installinitramfs` was not triggering `update-initramfs`, which is also fixed by adding an empty hook script.

closed 2025-06-24 by ivatet-amd