Xilinx-CNS/onload #293: Build fixes for 6.15 kernel

lukester1975, 2025-06-29:
Hello again

No doubt this is known, but just in case, this https://github.com/torvalds/linux/commit/e966ad0edd0056c7491b8f23992c11734ab61ddf in 6.15 breaks the build.

Trivial search and replace https://github.com/Xilinx-CNS/onload/compare/v9_0...lukester1975:onload:fix-6.15 builds and runs, but I'm not knowledgeable enough to say it's 100%...

Thanks!

ivatet-amd, 2025-06-30:
Thanks @lukester1975! Please stand by as this might take us a bit more time to address (and apply your patch).

boyracer2014, 2025-07-22:
looks like additional trouble (UNIXCB no longer exists?) 
```
  CC      src/lib/citools/hex_dump.o
src/lib/ciul/shrub_socket_kernel.c: In function ‘shrub_socket_get_files’:
src/lib/ciul/shrub_socket_kernel.c:113:8: error: implicit declaration of function ‘UNIXCB’ [-Werror=implicit-function-declaration]
  113 |   fp = UNIXCB(skb).fp;
      |        ^~~~~~
src/lib/ciul/shrub_socket_kernel.c:113:19: error: request for member ‘fp’ in something not a structure or union
  113 |   fp = UNIXCB(skb).fp;
      |                   ^
cc1: some warnings being treated as errors
make[6]: *** [/usr/src/kernels/6.15.7-1.el9.elrepo.x86_64/scripts/Makefile.build:203: src/lib/ciul/shrub_socket_kernel.o] Error 1
make[6]: *** Waiting for unfinished jobs....
  CC      src/lib/transport/ip/pio_buddy.o
  CC      src/lib/citools/hex_dump_to_raw.o
```